RobTillaart/I2C_EEPROM #4: updateByte

maxholgasson, 2020-12-21:
Would it be possible to add a function like in the internal Arduino lib for updating a byte? So read a byte and only write if it differs to safe some writing cycles of the eeprom?

RobTillaart, 2020-12-22:
Yes, definitely possible. Performance wise it might even be interesting. Will look into it later today.

RobTillaart, 2020-12-22:
Added **updateByte()** in 1.3.1.  Can you please verify it is working as I have no test hardware nearby.
There is a test example available that measures performance difference with **writeByte()**

maxholgasson, 2020-12-22:
Cool, will test it shortly and give you feedback.
I also started modifiying your lib to support trickle charging. I need to test it and verify if it's working, I can let you know, maybe you'd like to add it.

RobTillaart, 2020-12-22:
> started modifiying your lib to support trickle charging

Do you have a link to more info?

RobTillaart, 2020-12-22:
> Seems it's working

Then I'll close the issue, feel free to reopen if needed

closed 2020-12-22 by RobTillaart