RobTillaart/I2C_EEPROM #12: No Issue, but remarks/questions about Constructor parameter, and determineSize()

HJLK, 2021-01-07:
1) Constructor uses (default) I2C_EEPROM_PAGESIZE as "devicesize", which gives obviously not a right pagesize. Was this intentional?
2) In determineSize() when i equals 7 we have addr 65536, and 1 is added which overflows addr as a uint16_t and gives 1. 
Both "remarks" do not "influence" my 24c32. So determineSize() gives 4K correctly. But I try to understand the cpp, so could you please react?

RobTillaart, 2021-01-07:
@HJK

Questions are also welcome, 

The constant I2C_EEPROM_PAGESIZE  is in there for ages. A quick look showed it in a 2015 and I expect it in versions before that. Knowing myself it was probably the value matching the EEPROM I used for testing in those days.  So convenience for the author. It could be extended so it can be set compile time (see below) or even be removed so people have to make define it explicitly.

```cpp
#ifndef I2C_EEPROM_PAGESIZE 
#define I2C_EEPROM_PAGESIZE 64
#endif
```

**determineSize()** is indeed limited thereand should be updated to be able to work with larger EEPROMS.  
I will make an issue for that, so it is on the list and not forgotten.

HJLK, 2021-01-07:
For determineSize() I suggest to _subtract_ 1 in stead of adding 1 to addr. so you start with i=0 (511) and i=7 (65535).
As I use now the constructor with only the I2C address, _pagesize is wrong, and my EEPROM is not recognized. So I will use constructor _with correct size parameter from now on.

closed 2021-01-07 by HJLK