RobTillaart/I2C_EEPROM #13: DetermineSize() fails to detect larger EEPROMS

RobTillaart, 2021-01-07:
See remark in #12 

As the internal addr is a 16 bit the function will fail on 64K and larger EEPROMS.

RobTillaart, 2021-01-18:
Need to have test hardware e,g 64, 128 and maybe even 256 KB EEPROM
- 24LC512   64 KB
- 24LC1025 128 KB - other addressing scheme?
- AT24CM02 256 KB

RobTillaart, 2021-01-19:
Fixed in 1.3.2  - still waiting for different hardware to test.

**determineSize()** can detect up to 64KB EEPROMS.  
It is now also possible to determine 128, 256 and 512 byte sized EEPROMS - see readme.md

Larger EEPROMS need serious change in library to address beyond 64KB.
This will be investigated when hardware arrived.
For now the issue is closed and will be reopened if needed.

closed 2021-01-19 by RobTillaart

spcqike, 2023-03-03:
Hey,

have you got your hardware? i tried out an 24C256 today and the examples recognizes it as 32768 byte eeprom.

> I2C eeprom library: 1.7.1
> 
> isConnected:    1
> 
> TEST: determine size
> 80      FF
> 100     0
> 200     0
> 400     0
> 800     FF
> 1000    AA
> 2000    AA
> 4000    AA
> 8000    AA
> TIME: 71351
> SIZE: 32768 Bytes

is there anything one needs to change or do?

RobTillaart, 2023-03-03:
Hi,
Thanks for your questions

> have you got your hardware?

Yes, I bought a whole set off different sized EEPROM's and tested them.

> is there anything one needs to change or do?

No, 
As far as I know **determineSize()** works pretty well for more than 2 years now.
The **24C256** is a 256 kilobit EEPROM ==>  256/8 == 32 kiloByte == 32768 bytes. 
See https://ww1.microchip.com/downloads/en/devicedoc/doc0670.pdf

So looks good for me.

spcqike, 2023-03-03:
Well, thanks for your help. Of course you are right. I just mixed up bits and byte (or forgot that 256 is kb, not kB…) 

so yes, it works as expected :)

RobTillaart, 2023-03-03:
No problem, better ask if in doubt