RobTillaart/I2C_EEPROM #17: DetermineSize should work for 1 byte and 2 byte addresses

RobTillaart, 2021-01-26:
DetermineSize works but not for 1 and 2 bytes addresses in one go.

So the tests should be more explicit testing a given size.

RobTillaart, 2021-01-26:
Cause found

Three groups of address scheme
1. 24LC512, 24LC256, 24LC128, 24LC64, 24LC32* have a clean 2 byte addressing scheme  (*not tested)
2. 24LC16, 24LC08, 24LC04  use resp. 3,2,1 bit of the device address byte as memory address
3. 24LC02 and 24LC01 use 0 bit of the device address byte for memory, but they mask 3 bits in device address as don't care. 
Looks like a clean 1 byte address but in fact it is not.

I2C scanner sees in group 2 and 3 eight addresses on which the EEPROM responds.

Technically the 24LC16, 24LC08, 24LC04, 24LC02 and 24LC01 use all a similar addressing scheme.

Need to think about solution..

RobTillaart, 2021-01-27:
Currently implementing a new addressing scheme for all 24LCxx EEPROMS of 2K and below.
Did not to break existing code so far.
Testing will be finished later this week as I order some extra EEPROMS to test today.

owenduffy, 2021-01-27:
Thanks Rob... Owen.

RobTillaart, 2021-01-27:
@owenduffy 
in progress - https://github.com/RobTillaart/I2C_EEPROM/tree/develop

I have to do some tests before merging but it looks quite stable

RobTillaart, 2021-01-28:
Fixed in 1.4.0  2d60816

closed 2021-01-28 by RobTillaart