RobTillaart/I2C_EEPROM #21: Device Address not working

AndrejValand, 2021-01-29:
You can define the hardware address of an EEPROM, but then it uses the default 0x50. When using one address byte EEPROMS from ST or others (Example M24C08) that allow other addresses then 0x50 it does not work.

Fix to:
L:298 in I2C_eeprom.cpp 
uint8_t addr = _deviceAddress | ((memoryAddress >> 8) & 0x07);

RobTillaart, 2021-01-29:
Thanks for reporting. 
Which version of the library are you using?
In the 1.4.0 the addressing scheme was changed.

Will look into this lasap

RobTillaart, 2021-01-29:
I'm on it and need to check their datasheet.

RobTillaart, 2021-01-29:
The M24Cxx have no address lines, they have enable lines. 
The name sounds like they can be used for runtime address changing?
Or just interpret them as fixed address lines. 

- https://www.st.com/resource/en/datasheet/m24c16-w.pdf - Page 7
should work as it has no E2 line in its address

- https://www.st.com/resource/en/datasheet/m24c04-w.pdf  - Page 13
has an E1 and E2 enable line that can be interpreted as an address line.

- https://www.st.com/resource/en/datasheet/m24c08-w.pdf - Page 13
has an E2 enable line that can be interpreted as an address line.

- https://www.st.com/resource/en/datasheet/m24c01-w.pdf - Page 13 
has 3 enable lines.    (same for m24C02)

The following have a 2nd addressing scheme with the b4 bit set - something with identification page.
(do not have time to investigate that in detail soon)
- https://www.st.com/resource/en/datasheet/m24c32-w.pdf - Page 13
- https://www.st.com/resource/en/datasheet/m24c64-w.pdf - Page 14
- https://www.st.com/resource/en/datasheet/m24128-bw.pdf - Page 9
- https://www.st.com/resource/en/datasheet/m24256-bw.pdf - Page 9
- https://www.st.com/resource/en/datasheet/m24512-df.pdf - Page 9

1 and 2 Mbit EEPROM is not supported by this library (mentioned for completeness)
https://www.st.com/resource/en/datasheet/m24m01-df.pdf - Page 13
https://www.st.com/resource/en/datasheet/m24m02-dr.pdf - Page 13

AndrejValand, 2021-01-29:
I used V1.32 first. 

I have the E2 (Address/Enable) pin on a 24C08 wired to Vcc. With the old library it worked but i could use only 256bytes from 1024bytes of the chip, and determineSize() reported also 256bytes found. Then I changed to V1.4.0 and determineSize() stopped to work, I also had no access to my EEPROM anymore. After looking into I2C_eeprom.cpp I have seen when you address the EEPROM you use fixed 0x50 address not the one specified during initialization. Changing that to the line I have written above fixes the issue.

So in my case with E2 wired the hardware chip address is 0x54 not 0x50 anymore.

The E1 or E1,E2 lines are as you write hardware enable/address lines but when hardwiring it you also need send the correct address request when reading/writing the EEPROM. If its fixed to 0x50 it will just not work.

RobTillaart, 2021-01-29:
Thank you for explaining, I understood the problem however I have no ST hardware to test the solution.

Fix is in - https://github.com/RobTillaart/I2C_EEPROM/tree/develop

AndrejValand, 2021-01-29:
For M24C08 im using readByte, writeByte, readBlock, writeBlock, determineSize and isConnected work OK with this fix. Other function i did not check.

RobTillaart, 2021-01-29:
Working is good enough for me to merge :)
Thanks for pointing out that these "equivalents" were just a bit different.  This fix makes the library more useful!

If there are any other problems, just let me know !

commit 692f1a30 (2021-01-29): fix #21 addressing ST m24C08 et al + WireN support (#22)

* fix #21 addressing ST m24C08 et al + WireN support
closed 2021-01-29 by RobTillaart