RobTillaart/I2C_EEPROM #25: 512kbit eeprom problem

K0rkunc, 2021-05-01:
I am using 512 kbt eeprom but it sees 4 kb how can I fix it  

I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC512);

RobTillaart, 2021-05-01:
Hi,
I need more information to help you diagnose the problem.

- What is the type number on the EEPROM?
- Which platform do you use?
- Which version of the library?

- How do you conclude it only sees 4kbit?

K0rkunc, 2021-05-01:
24lc512  platformio arduino framework  1.4.2 lastest version

i did determine size  

[image]

ee.readBlock(40, (uint8_t *)ayopen, 5);
  ee.readBlock(46, (uint8_t *)ayclose, 5);
  ee.readBlock(82, (uint8_t *)blopen, 5);  i cant read this  
  ee.readBlock(58, (uint8_t *)blclose, 5);
  ee.readBlock(230, (uint8_t *)siopen, 5);
  ee.readBlock(70, (uint8_t *)siclose, 5);
  ee.readBlock(76, (uint8_t *)r1open, 5);
  ee.readBlock(82, (uint8_t *)r1close, 5);
  ee.readBlock(88, (uint8_t *)r2open, 5);
  ee.readBlock(196, (uint8_t *)r2close, 5);
  ee.readBlock(100, (uint8_t *)r3open, 5);
  ee.readBlock(210, (uint8_t *)r3close, 5);
  ee.readBlock(109, (uint8_t *)r4open, 5);
  ee.readBlock(115, (uint8_t *)r4close, 5);
  ee.readBlock(130, (uint8_t *)r5open, 5);
  ee.readBlock(216, (uint8_t *)r5close, 5);
  ee.readBlock(142, (uint8_t *)r6open, 5);
  ee.readBlock(148, (uint8_t *)r6close, 5);
  ee.readBlock(154, (uint8_t *)r7open, 5);
  ee.readBlock(160, (uint8_t *)r7close, 5);
  ee.readBlock(166, (uint8_t *)r8open, 5);
  ee.readBlock(172, (uint8_t *)r8close, 5);
  ee.readBlock(178, (uint8_t *)r9open, 5);
  ee.readBlock(184, (uint8_t *)r9close, 5);

  ay_i = ee.readByte(1);
  s470 = ee.readByte(2);
  s450 = ee.readByte(3);
  s440 = ee.readByte(4);
  s420 = ee.readByte(5);
  s400 = ee.readByte(6);
  s5000 = ee.readByte(7);
  s500 = ee.readByte(8);
  s600 = ee.readByte(9);
  ayenabled = ee.readByte(10);
  ayg = ee.readByte(11);
  ayr = ee.readByte(12);
  blg = ee.readByte(13);
  blr = ee.readByte(14);
  sienabled = ee.readByte(15);
  sig = ee.readByte(16);
  sir = ee.readByte(17);
  r1g = ee.readByte(18);
  r1r = ee.readByte(19);
  r2g = ee.readByte(20);
  r2r = ee.readByte(21);
  r3g = ee.readByte(22);
  r3r = ee.readByte(23);
  r4g = ee.readByte(24);
  r4r = ee.readByte(25);
  r5g = ee.readByte(26);
  r5r = ee.readByte(27);
  r6g = ee.readByte(28);
  r6r = ee.readByte(29);
  r7g = ee.readByte(30);
  r7r = ee.readByte(31);
  r8g = ee.readByte(32);
  r8r = ee.readByte(33);
  r9g = ee.readByte(34);
  r9r = ee.readByte(35);

K0rkunc, 2021-05-01:
http://ww1.microchip.com/downloads/en/devicedoc/21754m.pdf    

datasheet

RobTillaart, 2021-05-01:
So you read ~60 variables from EEPROM
- 35 are single byte values
- 24 are five bytes large

```
ee.readBlock(82, (uint8_t *)blopen, 5); i cant read this
```
What is the declaration of the blopen variable?
What are typical values?

**readBlock()** returns the number of bytes read. 
You should check in your code that it equals the requested nr of bytes.
That makes code robust. Do you do that?

RobTillaart, 2021-05-01:
Is the EEPROM by any chance write protected?  
dont know which pin it is

Do you use pull up resistors on the I2C bus?
which value, and how long is the wire?
Are there more I2C devices on the bus?

K0rkunc, 2021-05-01:
yes my charr length 5 byte 

now only one device is connected but later I will add 2 more devices to the same pins ..

I can save 8192 bytes to 512 kbit eeproma, but I can not. If possible, can you come to teamviewer please observe the error
597 023 776
iiz323

K0rkunc, 2021-05-01:
readblock length 5

K0rkunc, 2021-05-01:
can u connect me im waiting u on teamviewer 597 023 776
iiz323

K0rkunc, 2021-05-01:
if (request->argName(i) == "blopen");
    {
      strcpy(blopen, request->arg(i).c_str());
      ee.updateBlock(52, (uint8_t *)&blopen, 5);
    }

RobTillaart, 2021-05-01:
"blopen" is 6 bytes - you must store the trailing '\0' char too to be save.

What is the declaration of the blopen variable? 
please copy the line of code

K0rkunc, 2021-05-01:
all variables on read but only 2-3 variable cannot read or write  im tested 5 len

#include <Arduino.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <AsyncTCP.h>
#include <Adafruit_PWMServoDriver.h>
#include <Wire.h>
#include "SPIFFS.h"
#include "string.h"
#include "Hash.h"
#include "Wire.h"
#include "I2C_eeprom.h"

I2C_eeprom ee(0x50, 65535);
uint32_t start, dur1, dur2;
AsyncWebServer server(80);
Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver();

//ayarlar
char cihaz_adi[50] = {"TwinLED"}, cihaz_sifre[50] = {"passwords"};
int rf_sw = 1;

//test
uint8_t ay_i = 0, s470 = 0, s450 = 0, s440 = 0, s420 = 0, s400 = 0, s5000 = 0, s600 = 0, s500 = 0,
        ayg = 0, ayr = 0, ayenabled = 1, blg = 0, blr = 0, blenabled = 1, sig = 0, sir = 0, sienabled = 1,
        r1g = 0, r1r = 0, r2g = 0, r2r = 0, r3g = 0, r3r = 0, r4g = 0, r4r = 0, r5g = 0, r5r = 0,
        r6g = 0, r6r = 0, r7g = 0, r7r = 0, r8g = 0, r8r = 0, r9g = 0, r9r = 0;

char ayopen[9], ayclose[9], blopen[9], blclose[9], siopen[9], siclose[9], renk1_ac[6],
    r1open[9], r1close[9], r2open[9], r2close[9], r3open[9], r3close[9], r4open[9],
    r4close[9], r5open[9], r5close[9], r6open[9], r6close[9], r7open[9], r7close[9], r8open[9], r8close[9], r9open[9], r9close[9];

const char *ssid = "qwerty";
const char *password = "281218teo";

void notFound(AsyncWebServerRequest *request)
{
  request->send(404, "text/plain", "Elektronik ATOLYEM 0553 075 47 81");
}
void handleayarlar(AsyncWebServerRequest *request)
{
  request->send(200, "text/plain", String("ay=") + ay_i + String("&470=") + s470 + String("&400=") + s400 + String("&450=") + s450 + String("&440=") + s440 + String("&420=") + s420 + String("&5000k=") + s5000 + String("&500=") + s500 + String("&600=") + s600 + String("&setname=") + cihaz_adi + String("&setpassword=") + cihaz_sifre + String("&setrf=") + rf_sw + String("&ayopen=") + ayopen + String("&ayclose=") + ayclose + String("&blopen=") + String("&ayg=") + ayg + String("&ayr=") + ayr + blopen + String("&blclose=") + blclose + String("&blg=") + blg + String("&blr=") + blr + String("&siopen=") + siopen + String("&siclose=") + siclose + String("&sig=") + sig + String("&sir=") + sir + String("&blenabled=") + blenabled + String("&ayenabled=") + ayenabled + String("&sienabled=") + sienabled + String("&r1g=") + r1g + String("&r1r=") + r1r + String("&r1open=") + r1open + String("&r1close=") + r1close + String("&r2g=") + r2g + String("&r2r=") + r2r + String("&r2open=") + r2open + String("&r2close=") + r2close + String("&r3g=") + r3g + String("&r3r=") + r3r + String("&r3open=") + r3open + String("&r3close=") + r3close + String("&r4g=") + r4g + String("&r4r=") + r4r + String("&r4open=") + r4open + String("&r4close=") + r4close + String("&r5g=") + r5g + String("&r5r=") + r5r + String("&r5open=") + r5open + String("&r5close=") + r5close + String("&r6g=") + r6g + String("&r6r=") + r6r + String("&r6open=") + r6open + String("&r6close=") + r6close + String("&r7g=") + r7g + String("&r7r=") + r7r + String("&r7open=") + r7open + String("&r7close=") + r7close + String("&r8g=") + r8g + String("&r8r=") + r8r + String("&r8open=") + r8open + String("&r8close=") + r8close + String("&r9g=") + r9g + String("&r9r=") + r9r + String("&r9open=") + r9open + String("&r9close=") + r9close);
}

void handleayarlar_post(AsyncWebServerRequest *request)
{
  int args = request->args();
  for (int i = 0; i < args; i++)
  {
    Serial.printf("ARG[%s]: %s\n", request->argName(i).c_str(), request->arg(i).c_str());
    //test
    if (request->argName(i) == "loginpassword")
    {
      char temp[50];
      int durum;
      strcpy(temp, request->arg(i).c_str());
      Serial.println(temp);
      durum = strcmp(temp, cihaz_sifre);
      if (strcmp(temp, cihaz_sifre) == 0)
      {
        request->send(200, "text/plain", String("ay=") + ay_i + String("&470=") + s470 + String("&400=") + s400 + String("&450=") + s450 + String("&440=") + s440 + String("&420=") + s420 + String("&5000k=") + s5000 + String("&500=") + s500 + String("&600=") + s600 + String("&setname=") + cihaz_adi + String("&setpassword=") + cihaz_sifre + String("&setrf=") + rf_sw + String("&ayopen=") + ayopen + String("&ayclose=") + ayclose + String("&blopen=") + String("&ayg=") + ayg + String("&ayr=") + ayr + blopen + String("&blclose=") + blclose + String("&blg=") + blg + String("&blr=") + blr + String("&siopen=") + siopen + String("&siclose=") + siclose + String("&sig=") + sig + String("&sir=") + sir + String("&blenabled=") + blenabled + String("&ayenabled=") + ayenabled + String("&sienabled=") + sienabled + String("&r1g=") + r1g + String("&r1r=") + r1r + String("&r1open=") + r1open + String("&r1close=") + r1close + String("&r2g=") + r2g + String("&r2r=") + r2r + String("&r2open=") + r2open + String("&r2close=") + r2close + String("&r3g=") + r3g + String("&r3r=") + r3r + String("&r3open=") + r3open + String("&r3close=") + r3close + String("&r4g=") + r4g + String("&r4r=") + r4r + String("&r4open=") + r4open + String("&r4close=") + r4close + String("&r5g=") + r5g + String("&r5r=") + r5r + String("&r5open=") + r5open + String("&r5close=") + r5close + String("&r6g=") + r6g + String("&r6r=") + r6r + String("&r6open=") + r6open + String("&r6close=") + r6close + String("&r7g=") + r7g + String("&r7r=") + r7r + String("&r7open=") + r7open + String("&r7close=") + r7close + String("&r8g=") + r8g + String("&r8r=") + r8r + String("&r8open=") + r8open + String("&r8close=") + r8close + String("&r9g=") + r9g + String("&r9 [... cut ...]
  server.on("/script.js", HTTP_GET, [](AsyncWebServerRequest *request) { request->send(SPIFFS, "/script.js", "text/javascript"); });
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) { request->send(SPIFFS, "/index.html", "text/html"); });
  server.on("/api", HTTP_GET, handleayarlar);
  server.on("/api", HTTP_POST, handleayarlar_post);

  server.onNotFound(notFound);
  server.begin();
}

void loop(void)
{

  /*
    RtcDateTime now = Rtc.GetDateTime();
    printDateTime(now);
    Serial.println();
pwm.setPWM(0, 0, y+=100);
//delay(1000);
Serial.println(y);
if(y>4095){y=0;}*/
}

RobTillaart, 2021-05-01:
thanks I only neede the line

```cpp
char ... blopen[9], ...
```

So you have an array of nine bytes but fill only 5 of them from EEPROM.

what are a typical or expected values of the char array blopen?

I added code tags to your post for readability...

RobTillaart, 2021-05-01:
Had a quick look at the code and you need to learn how to use arrays.

e.g. you have repeating patterns like   rNg  and rNr

That would imply an array
```cpp
uint8_t rg[10];
uint8_t  rr[10];

// just 2 calls  to read all of them  or write them == faster and easier to maintain.

if ( ee.readBlock(baseAddress,RG rg, 10) != 10) ==> do error handling
if ( ee.readBlock(baseAddressRR, rr, 10) != 10) ...
```

RobTillaart, 2021-05-01:
Same for the rNopen and rN close

```cpp
char ropen[10][9];  // allocate 10 strings of length 9
char rclose[10][9];

if (ee.readBlock(baseAddressROPEN, (uint8_t *) ropen, 90) != 90) == > error
if (ee.readBlock(baseAddressRCLOSE, (uint8_t *) ropen, 90) != 90) == > error
```

RobTillaart, 2021-05-01:
BUG

ee.readBlock(82, (uint8_t *)blopen, 5); i cant read this
...
ee.readBlock(82, (uint8_t *)r1close, 5);

blopen is probably overwritten by r1close

Another reason to go for the array solution, you need less administration.

K0rkunc, 2021-05-01:
I think the problem is not about using arrays, but about saving and retrieving data.

RobTillaart, 2021-05-01:
Agree, arrays is a way to optimize readability and maintainability of code.

my question is still open regarding the string blopen

blopen an array of nine bytes but you fill only 5 of them from EEPROM.
what are a typical or expected values of the char array blopen?

RobTillaart, 2021-05-02:
> I am using 512 kbt eeprom but it sees 4 kb how can I fix it
> I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC512);

Q: Does your application not work correctly because of the result of the **determineSize()** function?
NB the function does not appear anywhere in the code?

### code review

I did a code review of the whole library this morning and found no path in the code that would allow the **determineSize()** function to return 4kbit when a 512kbit EEPROM is connected. From your information it looks like the right parameter I2C_DEVICESIZE_24LC512 was given. 

That does not mean your observation is invalid, only that I cannot explain it from the code. 
All EEPROM sizes have been tested to work by myself (Arduino UNO = 5V, 100KHz, 4K7 pullup, and some with ESP32) except the 24LC32. 

Therefor the problem analysis must focus on your hardware as I cannot replicate the problem.

### questions hardware

Q1: what board do you use? 5V or 3V3 ?
(3V3 is more sensitive to power glitches)

Q2: have you measured the actual voltage of the EEPROM?
(A common problem with some 3V3 devices is that they have 3V1 or 3V2)

Q3: have you tried another power supply?
(some cheaper ones do not have a flat voltage, an extra capacitor could help)

Q4: if a 3V3 device, did you connect the Vcc of the EEPROM to 5V or to 3V3?
(the reference signals LOW / HIGH are defined by Vcc) 

Q5: what pull ups do you use on the I2C SDA and SCL line?
(smaller values improve the signal by making them more "square")

Q6: do you get the 4kb readout consequently?
(does it happen 20 out of 20, or less)

Q7: have you tried another EEPROM? 
Same size or other size? 
Do you see the same error?

K0rkunc, 2021-05-02:
im solved ds1307 eeprom adress 0x50 crashed im change address and solved thank u so much

RobTillaart, 2021-05-02:
Good to hear it is solved!

closed 2021-05-02 by RobTillaart