RobTillaart/I2C_EEPROM #31: read errors

K0rkunc, 2021-07-26:
only this variable gives �������D

I2C_eeprom ee(0x51, I2C_DEVICESIZE_24LC512);

 if (request->argName(i) == "setname")
    {
      if (strcmp(cihaz_adi, request->arg(i).c_str()) != 0)
      {
        strcpy(cihaz_adi, request->arg(i).c_str());
        pass_len1 = strlen(cihaz_adi);
        ee.writeBlock(670, (uint8_t *)&cihaz_adi, sizeof(pass_len1));
        ee.writeByte(350, pass_len1);
      }
    }

 **Readblock**

> pass_len = ee.readByte(290);
>   pass_len1 = ee.readByte(350);
>   ee.readBlock(40, (uint8_t *)ayopen, 5);
>   ee.readBlock(50, (uint8_t *)ayclose, 5);
>   ee.readBlock(280, (uint8_t *)blopen, 5);
>   ee.readBlock(70, (uint8_t *)blclose, 5);

   ee.readBlock(80, (uint8_t *)siopen, 5);
  ee.readBlock(90, (uint8_t *)siclose, 5);
  ee.readBlock(100, (uint8_t *)r1open, 5);
  ee.readBlock(110, (uint8_t *)r1close, 5);
  ee.readBlock(120, (uint8_t *)r2open, 5);
  ee.readBlock(130, (uint8_t *)r2close, 5);
  ee.readBlock(140, (uint8_t *)r3open, 5);
  ee.readBlock(150, (uint8_t *)r3close, 5);
  ee.readBlock(160, (uint8_t *)r4open, 5);
  ee.readBlock(170, (uint8_t *)r4close, 5);
  ee.readBlock(180, (uint8_t *)r5open, 5);
  ee.readBlock(190, (uint8_t *)r5close, 5);
  ee.readBlock(200, (uint8_t *)r6open, 5);
  ee.readBlock(210, (uint8_t *)r6close, 5);
  ee.readBlock(220, (uint8_t *)r7open, 5);
  ee.readBlock(230, (uint8_t *)r7close, 5);
  ee.readBlock(240, (uint8_t *)r8open, 5);
  ee.readBlock(250, (uint8_t *)r8close, 5);
  ee.readBlock(260, (uint8_t *)r9open, 5);
  ee.readBlock(270, (uint8_t *)r9close, 5);
  ee.readBlock(300, (uint8_t *)cihaz_sifre, pass_len);
  ee.readBlock(670, (uint8_t *)cihaz_adi, pass_len1);
  ee.readBlock(455, (uint8_t *)wifi_name, 17);
  ee.readBlock(500, (uint8_t *)kapat_time[0], 5);
  ee.readBlock(500, (uint8_t *)kapat_time[0], 5);
  ee.readBlock(506, (uint8_t *)kapat_time[1], 5);
  ee.readBlock(513, (uint8_t *)kapat_time[2], 5);
  ee.readBlock(519, (uint8_t *)kapat_time[3], 5);
  ee.readBlock(525, (uint8_t *)kapat_time[4], 5);
  ee.readBlock(531, (uint8_t *)kapat_time[5], 5);
  ee.readBlock(537, (uint8_t *)kapat_time[6], 5);
  ee.readBlock(543, (uint8_t *)kapat_time[7], 5);
  //Serial.println(cihaz_sifre);
  //ay_i = ee.readByte(1);
  //s470 = ee.readByte(2);
  //s450 = ee.readByte(3);
  //s440 = ee.readByte(4);
  //s420 = ee.readByte(5);
  //s400 = ee.readByte(6);
  //s5000 = ee.readByte(7);
  //s500 = ee.readByte(8);
  // s600 = ee.readByte(9);
  ayenabled = ee.readByte(10);
  ayg = ee.readByte(11);
  ayr = ee.readByte(12);
  blg = ee.readByte(13);
  blr = ee.readByte(14);
  sienabled = ee.readByte(15);
  blenabled = ee.readByte(36);
  sig = ee.readByte(16);
  sir = ee.readByte(17);
  r1g = ee.readByte(18);
  r1r = ee.readByte(19);
  r2g = ee.readByte(20);
  r2r = ee.readByte(21);
  r3g = ee.readByte(22);
  r3r = ee.readByte(23);
  r4g = ee.readByte(24);
  r4r = ee.readByte(25);
  r5g = ee.readByte(26);
  r5r = ee.readByte(27);
  r6g = ee.readByte(28);
  r6r = ee.readByte(29);
  r7g = ee.readByte(30);
  r7r = ee.readByte(31);
  r8g = ee.readByte(32);
  r8r = ee.readByte(33);
  r9g = ee.readByte(34);
  r9r = ee.readByte(35);
  rf_en = ee.readByte(37);
  sn_i = ee.readByte(420);
  dk_i = ee.readByte(421);
  saat_i = ee.readByte(422);
  gun_i = ee.readByte(423);
  ays_i = ee.readByte(424);

RobTillaart, 2021-07-27:
I do not understand the problem you have. 
You make a statement about "only this variable gives �������D"
and provides some uncompilable code.

Please provide a minimal sized sketch that shows the problem.

K0rkunc, 2021-07-27:
**Hello**

**I am writing eeprom a char variable in uint8_t format as in the codes below.**

**then I read this data at boot and it gives irrelevant values.**

**I don't have this problem in other reading and writing areas, but I'm having this problem in "device_name" variable, **

**please can you help me with this?**

**I get the device name from the esp32 async web server interface and save it.**

**by the way, the recording function returns a value of 4.**

if (request->argName(i) == "setname")
    {
      if (strcmp(cihaz_adi, request->arg(i).c_str()) != 0)
      {
        snprintf(cihaz_adi,sizeof(cihaz_adi),"%c",request->arg(i).c_str());
        pass_len1 = strlen(cihaz_adi);
        ee.writeBlock(600, (uint8_t *)&cihaz_adi, 4);
        ee.writeByte(350, pass_len1);
      }
    }

**then I try to read this value at device startup.**

pass_len1 = ee.readByte(350);
ee.readBlock(600, (uint8_t *)cihaz_adi, pass_len1);

but this is how the value I read comes in. �������

RobTillaart, 2021-07-27:
> Please provide a minimal sized sketch that shows the problem.

Are you sure that 
- there is a value written to EEPROM?
- it is the right value?

RobTillaart, 2021-07-27:
> by the way, the recording function returns a value of 4.

such remark does not tell me anything if I do not have a complete minimal sketch to verify

K0rkunc, 2021-07-27:
full project codes https://www.paste.tc/code-84118

yes after transferring the values ​​I get from the interface to the char variable

I am reading with serial print the value is correct

but unfortunately when i save and restart, bad value comes

RobTillaart, 2021-07-27:
> minimal sketch

...

K0rkunc, 2021-07-27:
#include "I2C_eeprom.h"

char cihaz_adi[50]={"TwinsLED"};

I2C_eeprom ee(0x51, I2C_DEVICESIZE_24LC512);

Setup()
{
 Wire.begin();
  Wire.setClock(400000);
ee.begin();

 

strcpy(cihaz_adi, request->arg(i).c_str());
        pass_len1 = strlen(cihaz_adi);
        cihaz_adi[5]='\0';
        ee.writeBlock(600, (uint8_t *)&cihaz_adi, 4);
        ee.writeByte(350, pass_len1);

pass_len1 = ee.readByte(350);
ee.readBlock(600, (uint8_t *)cihaz_adi, pass_len1);

}

RobTillaart, 2021-07-27:
**Not tested code**, no time today to elaborate on this, but should get you started how to store a text in EEPROM

```cpp
//    FILE: I2C_eeprom_store_text.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo I2C_EEPROM library

// uses a 24LC256 (32KB) EEPROM
// might need adaptions for other EEPROMS (page size etc)

#include "Wire.h"
#include "I2C_eeprom.h"

I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC256);

void setup()
{
  Serial.begin(115200);
  while (!Serial);      // wait for Serial port to connect.
  Serial.println(__FILE__);

  ee.begin();
  if (! ee.isConnected())
  {
    Serial.println("ERROR: Can't find eeprom\nstopped...");
    while (1);
  }

  Serial.print("I2C_EEPROM_VERSION: ");
  Serial.print(I2C_EEPROM_VERSION);
  Serial.println("\n");

  Serial.println("\nwrite string");
  ee.setBlock(600, 0, 128);
  dumpEEPROM(600, 128);
  char data[] = "hello world";
  ee.writeBlock(600, (uint8_t*) data, strlen(data) + 1);  // add the \0 char to be sure   <<<<<<<<<<<<<<<
  dumpEEPROM(600, 128);

  Serial.println("\nread string");
  char dataRead[100];
  memset(dataRead, 0, 100);
  ee.readBlock(600, (uint8_t*)dataRead, 12);
  Serial.println(dataRead);

  Serial.println("\tDone...");
}

void loop()
{
}

void dumpEEPROM(uint16_t memoryAddress, uint16_t length)
{
  const int BLOCK_TO_LENGTH = 10;
  
  Serial.print("\t  ");
  for (int x = 0; x < 10; x++)
  {
    if (x != 0) Serial.print("    ");
    Serial.print(x);
  }
  Serial.println();

  // block to defined length
  memoryAddress = memoryAddress / BLOCK_TO_LENGTH * BLOCK_TO_LENGTH;
  length = (length + BLOCK_TO_LENGTH - 1) / BLOCK_TO_LENGTH * BLOCK_TO_LENGTH;

  byte b = ee.readByte(memoryAddress);
  for (unsigned int i = 0; i < length; i++)
  {
    char buf[6];
    if (memoryAddress % BLOCK_TO_LENGTH == 0)
    {
      if (i != 0) Serial.println();
      sprintf(buf, "%05d", memoryAddress);
      Serial.print(buf);
      Serial.print(":\t");
    }
    sprintf(buf, "%03d", b);
    Serial.print(buf);
    b = ee.readByte(++memoryAddress);
    Serial.print("  ");
  }
  Serial.println();
}

// END OF FILE
```

RobTillaart, 2021-07-27:
> #include "I2C_eeprom.h"
>
> char cihaz_adi[50]={"TwinsLED"};
>
> I2C_eeprom ee(0x51, I2C_DEVICESIZE_24LC512);
>
> Setup()
> {
> Wire.begin();
> Wire.setClock(400000);
> ee.begin();
>
> strcpy(cihaz_adi, [...]

Does not compile

K0rkunc, 2021-07-27:
Why

RobTillaart, 2021-08-16:
> Why

It does not compile because the code you posted is an incomplete half sketch.

Please provide a minimal sketch that shows the problem. 
(or if it is solved you may close the issue)

RobTillaart, 2021-08-24:
As there is no answer I assume the problem is solved, therefor I close the issue. If the issue still exists you can reopen the issue, or better ask on https://forum.arduino.cc/ for help on your sketch as your problem does not appear to be a library problem.

closed 2021-08-24 by RobTillaart