RobTillaart/I2C_EEPROM #32: Possible bug in EEPROM format example

djcaf, 2021-07-29:
I have only just come across this library so I may have misunderstood.

However in one of the examples;  I2C_EEPROM/examples/I2C_eeprom_format/I2C_eeprom_format.ino 
it looks like there may be a bug. I assume this is suppose to write a fixed value to every address on the EEPROM. The following loop looks suspicious,

for (uint32_t i = 0; i < size; i += 128)
    {
      if (i % 1024 == 0) Serial.print('.');
      ee.setBlock(0, 0xFF, 128);
    }

Assuming I am understanding it correctly this writes the value 0xFF to 128 places always starting at address 0.

I would assume the line should be:

ee.setBlock(i, 0xFF, 128);

Apologies if I have misunderstood something, I have not had a chance to look at this library in detail yet.

RobTillaart, 2021-07-29:
You caught a real bug!

RobTillaart, 2021-07-29:
Fixed the example in master,
I will not create a new release for this as it is not in the core.

djcaf, 2021-07-29:
Cool, I have made that same sort of bug myself many times in the past :-)

closed 2021-07-29 by RobTillaart