RobTillaart/I2C_EEPROM #33: 24LC32A

damarcrazy, 2021-08-22:
Best regard. Please I want to ask you for help to be able to upload a matrix of pointers using an arduino uno with an eeprom 24LC32A or larger and then display it on LCD
Thanks in advance for your support.

#include <I2C_eeprom.h>
#include "Arduino.h"
#include <Wire.h>                                     // libreria de comunicacion por I2C
#include <LiquidCrystal_I2C.h>                        // libreria para LCD por I2C

I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC32);

uint32_t start, diff;

/***********************************************************
   Base de datos de palabras y frases
 ***********************************************************/
const char* palabras[] = {
  "Maquillarse",
  "Playa",
  "Erizo",
  "Macarena",
  "Cirujano",
  "Huracan",
  "Facebook",
  "Estreñimiento",
  "Accidente de coche",
  "Criminal",
  "El presidente",
  "Chivato",
  "Club de lucha",
  "Hipo",
  "Jefe",
  "Detective",
  "Ipad",
  "Hacer galletas",
  "Chiste",
  "Puta",
  "Peluca",
  "Oveja negra",
  "Avion",
  "Macho",
  "Bebe",
  "Rayo de sol",
  "Brindis",
  "Limbo",
  "Cuchara",
  "Sorpresa",
  "Claque",
  "Bañar a un gato",
  "Cambiar pañal",
  "Pasear al perro",
  "Levantar pesas",
  "Jugar a las cartas",
};

/***********************************************************
   Inicializacion y funcion principal
 ***********************************************************/

LiquidCrystal_I2C lcd(0x27, 16, 2);                   //Crear el objeto lcd  direccion  0x27 y 16 columnas X 2 filas

void setup() {
  lcd.init();
  lcd.backlight();
  lcd.clear();
  pinMode(3, INPUT_PULLUP);
  ee.begin();
  if (! ee.isConnected())
  {
    lcd.setCursor(0,0); lcd.print(F("ERROR:No hay acceso con la memeoria..."));
    while (1);
  }

  ee.writeBlock(0, (uint8_t *) &palabras, 50);
  //ee.writeBlock(60, (uint8_t *) &data3, sizeof(data3));
 // dumpEEPROM(0, 128);

}

void loop() {
  
}

RobTillaart, 2021-08-22:
updated your post with syntax highlighting => readability

will be later this week before I have time to dive into this problem.

tIP: You need to make a for loop and go over all the entries of the palabras table.

RobTillaart, 2021-08-22:
@damarcrazy 
I assume the goal is to write the 40 or so strings into EEPROM first, is that correct?

Storing run time pointers in EEPROM makes no sense as in theory  and practice there is no guarantee that the 
data will be on the same addresses.

damarcrazy, 2021-08-22:
Thanks for your suggestion I will work on that while you can help me happy
weekend

RobTillaart, 2021-08-25:
searching for the n-th entry looks something like 

```cpp

void search(uint8_t n)
{
  if (n > 36) return;

  // find n-th entry
  uint32_t position = 0;
  for (uint8_t i = 0; i < n; i++) position += ee.readByte(position);

  // read and print it
  char buffer[40];
  uint8_t length = ee.readByte(position);
  ee.readBlock(position, (uint8_t *) &buffer, length);
  Serial.print(n);
  Serial.print("\t");
  Serial.println(buffer);
}
```

I have no eeprom nearby to test, :)

RobTillaart, 2021-08-29:
Any progress made?

damarcrazy, 2021-08-29:
Cordial greetings, sorry for not being able to respond earlier, I was
traveling today, I will test the recommendations you gave me. Thanks

RobTillaart, 2021-09-12:
Any progress?

RobTillaart, 2021-09-20:
As there is no progress I close this issue. Please reopen if needed.

closed 2021-09-20 by RobTillaart