RobTillaart/I2C_EEPROM #38: AT24C128C can i use this one?

daeynasvistas, 2022-02-03:
can i use this one AT24C128C with your lib?

all seems to go well but if i use your example:
**I2C_eeprom_struct.ino**

when read value i get:

```
size: 	12
write: 	1626
read: 	1708
temp:	0.00
hum:	37.01
pres:	0.01

done...

```

**using:**

```
  // make measurement here
  measurement.temperature = 22.5;
  measurement.humidity    = 53.1;
  measurement.pressure    = 1000.9;
```

RobTillaart, 2022-02-03:
Never tested it with a  AT24C **128** C, so its memory mapping might be different.
Think the **128** needs at least some constants (paging etc) defined to get it working. 

**theoretical approach**
You should check the datasheet if its memory mapping matches one of the types.

**pragmatic approach**
My first try would be to use it as a **24LC64** which seems smaller.

RobTillaart, 2022-02-06:
Have you adapted this line:
```cpp
I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC256);
```
to
```cpp
I2C_eeprom ee(0x50, I2C_PAGESIZE_24LC128);
```

daeynasvistas, 2022-02-07:
Yes i did.

`I2C_eeprom ee(0x51, I2C_DEVICESIZE_24LC128);
`

i2c scanner find it on 0x51

maybe its on write protection mode.. i didnt ground the WP pin

> WRITE PROTECT (WP): The write protect input, when connected to GND, allows normal write
> operations. When WP is connected high to VCC, all write operations to the memory are inhibited. If the pin is left floating, the WP pin will be internally pulled down to GND if the capacitive
> coupling to the circuit board VCC plane is <3 pF. If coupling is >3 pF, Atmel recommends connecting the pin to GND.

RobTillaart, 2022-02-07:
When WP is floating strange things can happen...

RobTillaart, 2022-02-08:
@daeynasvistas 

> maybe its on write protection mode.. i didnt ground the WP pin

did it work?

daeynasvistas, 2022-02-09:
sorry didnt reply .. i will try to ground WP and reply to you when done

RobTillaart, 2022-02-21:
Any progress?

daeynasvistas, 2022-02-22:
yes..its working! 

```
VERSION: 1.5.2

size: 	12
write: 	3384
read: 	1709
temp:	22.50
hum:	53.10
pres:	1000.90

done...

```

i use this one,  
http://ww1.microchip.com/downloads/en/devicedoc/atmel-8734-seeprom-at24c128c-datasheet.pdf

on custom ESP32 pico board:

with 
`I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC128);`

RobTillaart, 2022-02-22:
Hooray, another happy customer 👍
I close the issue, thanks for testing

closed 2022-02-22 by RobTillaart