RobTillaart/I2C_EEPROM #48: Suspected crash when calling begin()

djcaf, 2022-12-02:
Had a strange issue with code suddenly not running giving an USB device error in Windows and uploads no longer possible via USB. I have replicated (mostly) with the code below using the EEPROM library. Windows does not give an USB device error with this code (maybe due to delay() in my test code which delays the point at which the device fails). However uploads still are no longer possible when running this code so I guess the CPU is still crashing.

What should happen is after stat or upload the builtin LED is off for 3 seconds then comes on. What happens is the LED does not come on and USB uploads do not work. Pressing reset twice in quick succession allows new code to be uploaded.

It is begin() that fails and from what I could tell _wire is nullptr in the begin() code for some reason - does not look possile though.

It works (may just hide the issue) when I use the I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC256, &Wire) line instead. My first thought is undefined behaviour somewhere, but I have eliminated my code so that can't be the cause. I do not have a spare Arduino with me at the moment to test. The Arduino is only connected to a EEPROM chip but the same issue occurs when completely disconnected (i.e. the Arduino is removed from the breadboard).

Board in question is the Nano 33 IoT and the IDE is 2.0.2
The RaspberryPi PICO does the same thing. It flashes a code with its built in LED as well - Cannot find what my specific pattern means but essentially means it has crashed also which probably eliminates my hardware as a cause.

Thanks.

#include "Wire.h"
#include "I2C_eeprom.h"

I2C_eeprom ee(0x50, &Wire); //Does not work
//I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC256, &Wire); //Works

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  delay(3000);
  ee.begin();
  digitalWrite(LED_BUILTIN, HIGH);
}

void loop() {
  // put your main code here, to run repeatedly:

}

RobTillaart, 2022-12-02:
Thanks for this issue, I'm quite busy and it might take some time before investigating.

djcaf, 2022-12-02:
> updated your post for syntax highlighting (is quite simple have a look)

Thanks. I don't log things in git hub often I will remember in the future :-)

RobTillaart, 2022-12-02:
now the constructor calls the other one.
If you place the (appropriate) code in the first constructor does it work?  

```cpp
I2C_eeprom::I2C_eeprom(const uint8_t deviceAddress, TwoWire * wire)
{
    _deviceAddress = deviceAddress;
    _deviceSize = I2C_PAGESIZE_24LC256;
    _pageSize = getPageSize(_deviceSize);
    _wire = wire;

    //  Chips 16Kbit (2048 Bytes) or smaller only have one-word addresses.
    this->_isAddressSizeTwoWords = _deviceSize > I2C_DEVICESIZE_24LC16;
}
```

RobTillaart, 2022-12-03:
Thanks for testing, I will create a PR later today.

RobTillaart, 2022-12-03:
Arduino UNO gives the same problem, 
Checked the examples of the lib and they all use the 2nd constructor (also in my projects)
Good catch!

closed 2022-12-03 by RobTillaart

RobTillaart, 2022-12-03:
Merged PR,
Again thanks for catching the bug, appreciated!

commit a81db30e (2022-12-03): fix #48
commit 8f530b6d (2022-12-03): fix #48 constructor (#49)

* fix #48