RobTillaart/I2C_EEPROM #50: writeByte / readByte

4r53N1ck, 2023-01-11:
Hello! I have an issue (or two, I can't tell, really) with this library: readByte function returns only 0x000 on the serial monitor. I don't know if isn't writing, reading or none. I let you the sample code. 

Note: Since I didn't know if my code was bad or not, I tride your other library (I2C_24LC1025) and these functions works on that one.

```
#include <Wire.h>
#include <I2C_eeprom.h>

uint32_t eepromSize;
uint8_t b, c, d;

#define SERIAL_OUT Serial

I2C_eeprom ee(0x50);

void setup(){
  SERIAL_OUT.begin(9600);
  ee.begin();
  if(!ee.isConnected()){
    SERIAL_OUT.println("No memory detected. Freezing.");
    while(1);
  }
  SERIAL_OUT.println("Memory detected!");
  delay(10);
  SERIAL_OUT.print("Memory size in bytes: ");
  eepromSize = ee.determineSize();
  SERIAL_OUT.print(eepromSize);
  SERIAL_OUT.print('\n');
  ee.writeByte(0x0000,0xA1);
  ee.writeByte(0x0001,0xB2);
  ee.writeByte(0x0002,0xC3);
  SERIAL_OUT.print("Your written data is: 0h");
  SERIAL_OUT.print(ee.readByte(0x0000),HEX);
  SERIAL_OUT.print(ee.readByte(0x0001),HEX);
  SERIAL_OUT.print(ee.readByte(0x0002),HEX);
  SERIAL_OUT.print('\n');
}

void loop(){
  
}
```

RobTillaart, 2023-01-11:
Thanks for the issue, I will look into it tomorrow

RobTillaart, 2023-01-11:
Which EEPROM type do you use?
Which board do you use?
Do you have pull up resistors on the I2C lines?

RobTillaart, 2023-01-11:
Can you post the output of the sketch sbove?

4r53N1ck, 2023-01-11:
EEPROM = AT24C32 (the one that comes with the DS1307 module)
Board = Arduino NANO (no clone)
Yes, I do have pull up resistors on the I2C lines (included in the module).

Output of the sketch:
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\4r53n1c\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\4r53n1c\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\4r53n1c\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\4r53n1c\AppData\Local\Temp\arduino_build_759683 -warnings=none -build-cache C:\Users\4r53n1c\AppData\Local\Temp\arduino_cache_604481 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose C:\Users\4r53n1c\Desktop\PruebaEEPROM\PruebaEEPROM\PruebaEEPROM.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\4r53n1c\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\4r53n1c\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\4r53n1c\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\4r53n1c\AppData\Local\Temp\arduino_build_759683 -warnings=none -build-cache C:\Users\4r53n1c\AppData\Local\Temp\arduino_cache_604481 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose C:\Users\4r53n1c\Desktop\PruebaEEPROM\PruebaEEPROM\PruebaEEPROM.ino
Using board 'nano' from platform in folder: C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\4r53n1c\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Detecting libraries used...
"C:\\Users\\4r53n1c\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\4r53n1c\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\4r53n1c\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\eightanaloginputs" "C:\\Users\\4r53n1c\\AppData\\Local\\Temp\\arduino_build_759683\\sketch\\PruebaEEPROM.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
"C:\\Users\\4r53n1c\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\4r53n1c\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\4r53n1c\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\eightanaloginputs" "-IC:\\Users\\4r53n1c\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\Wire\\src" "C:\\Users\\4r53n1c\\AppData\\Local\\Temp\\arduino_build_759683\\sketch\\PruebaEEPROM.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for I2C_eeprom.h: [I2C_EEPROM@1.7.0]
ResolveLibrary(I2C_eeprom.h)
  -> candidates: [I2C_EEPROM@1.7.0]
[... cut ...]
avrdude: load data flash data from input file C:\Users\4r53n1c\AppData\Local\Temp\arduino_build_759683/PruebaEEPROM.ino.hex:
avrdude: input file C:\Users\4r53n1c\AppData\Local\Temp\arduino_build_759683/PruebaEEPROM.ino.hex contains 5252 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.70s

avrdude: verifying ...
avrdude: 5252 bytes of flash verified

avrdude done.  Thank you.
```
Output on the serial monitor:
```
Memory detected!
Memory size in bytes: 4096
Your written data is: 0h000
```

RobTillaart, 2023-01-12:
Quick check 
- latest version of the library 1.7.0
- correctly determines the size of the AT24C32 (32Kb == 4KB == 4096)  !! this uses read / write under the hood.
- the value read is 0 for all three bytes, which is not expected.

Can you try this version, to check if there is an I2C error when writing?
```cpp
#include <Wire.h>
#include <I2C_eeprom.h>

uint32_t eepromSize;
uint8_t b, c, d;

#define SERIAL_OUT Serial

I2C_eeprom ee(0x50);

void setup(){
  SERIAL_OUT.begin(9600);
  ee.begin();
  if(!ee.isConnected()){
    SERIAL_OUT.println("No memory detected. Freezing.");
    while(1);
  }
  SERIAL_OUT.println("Memory detected!");
  delay(10);
  SERIAL_OUT.print("Memory size in bytes: ");
  eepromSize = ee.determineSize();
  SERIAL_OUT.print(eepromSize);
  SERIAL_OUT.print('\n');

  SERIAL_OUT.println("\nWriteByte returns:");
  SERIAL_OUT.println(ee.writeByte(0x0000,0xA1), HEX);
  SERIAL_OUT.println(ee.writeByte(0x0001,0xB2), HEX);
  SERIAL_OUT.println(ee.writeByte(0x0002,0xC3), HEX);

  SERIAL_OUT.print("\nYour written data is: 0h");
  SERIAL_OUT.print(ee.readByte(0x0000),HEX);
  SERIAL_OUT.print(ee.readByte(0x0001),HEX);
  SERIAL_OUT.print(ee.readByte(0x0002),HEX);
  SERIAL_OUT.print('\n');
}

void loop(){
}
```

4r53N1ck, 2023-01-12:
Ok, wait a second.

4r53N1ck, 2023-01-12:
This is the output:

Memory detected!
Memory size in bytes: 4096

WriteByte returns:
0
0
0

Your written data is: 0h000

RobTillaart, 2023-01-12:
datasheet - https://ww1.microchip.com/downloads/en/DeviceDoc/doc0336.pdf

found this in the top of the datasheet
- **Self-Timed Write Cycle (10 ms max)**
The library uses a write delay of 5 ms.

please add a call after the isConnected() part
```cpp
ee.setExtraWriteCycleTime(6000);
```

4r53N1ck, 2023-01-12:
Added. Still no luck.

RobTillaart, 2023-01-12:
> This is the output:
> 
> Memory detected! Memory size in bytes: 4096
> 
> WriteByte returns: 0 0 0
> 
> Your written data is: 0h000

So there is no I2C error reported.

4r53N1ck, 2023-01-12:
Correct. Furthermore, I've tried your "writeBlock" function successfully (returns all 0 values which means it writes OK).

RobTillaart, 2023-01-12:
mmm, still have the feeling that timing is the cause.

The return of 0 means that I2C handshake is OK, it is still no 100% guarantee data is actually written. 

You stated that **I2C_24LC1025** library did work correctly. 
Can you try to read the 3 EEPROM addresses with **I2C_24LC1025** after you have written them with **I2C_EEPROM** lib
(be sure you do not overwrite when reading)

That should provide information if the data is written by the **I2C_EEPROM** lib

4r53N1ck, 2023-01-12:
OK, let me try that. Give me a moment.

4r53N1ck, 2023-01-12:
Confirmed: I2C_eeprom.h is not writing on the EEPROM.

RobTillaart, 2023-01-12:
> Confirmed: I2C_eeprom.h is not writing on the EEPROM.

OK, that makes it strange that the determineSize() does work as it uses writeByte too.

Lets go "extreme", 
can you put a delay(50) before every write() so it has enough time to do its internal thingies.

4r53N1ck, 2023-01-12:
I've tried the following:
Format EEPROM with I2C_24LC1025 library (all bytes should be 0xFF, right?).
Write 0xA1, 0xB2 and 0xC3 at 0, 1 and 2 EEPROM Addresses with I2C_eeprom library.
Read 0, 1 and 2 Addresses with I2C_24LC1025 and shows 0h000 again.

RobTillaart, 2023-01-12:
> Format EEPROM with I2C_24LC1025 library (all bytes should be 0xFF, right?).

no ==>  ee.setBlock(i, **0x00**, 128);

RobTillaart, 2023-01-12:
How is the WP (writeProtect) pin connected?

4r53N1ck, 2023-01-12:
WP is grounded on this DS1307 module.

RobTillaart, 2023-01-12:
Can you confirm that GND with a DMM? (multimeter)

4r53N1ck, 2023-01-12:
Confirmed WP pin is grounded with multimeter.

Let me try format, write and read, only this time setting bytes with 0xFF to see if there is some changes. I'm suspecting that "readByte" function returns a confirmation byte.

RobTillaart, 2023-01-12:
Note: Page 5 datasheet states that for the 1.8V version the writeCycleTime can even be 20 ms.

4r53N1ck, 2023-01-12:
This is what I've got so far:
Formatted with 0xFF values with your I2C_24LC1025-> OK
Read EEPROM at 0, 1 and 2 addresses with same library -> OK (Throws 0hFFFFFF on serial monitor)
Write EEPROM at 0, 1 and 2 addresses with same library-> OK (Tried reading same addresses with same library and throws 0xA1B2C3 on serial monitor)
Read EEPROM at 0, 1 and 2 addresses with I2C_eeprom library -> FAIL (Throws 0hFFFFFF on the serial monitor)

I'll make another test, formatting memory, writing it with I2C_eeprom and reading with I2C_24LC1025

4r53N1ck, 2023-01-12:
Confirmed: I2C_eeprom is not writing. I've changed A1B2C3 for AABBCC and still reads A1B2C3

RobTillaart, 2023-01-12:
Can you try this code
```cpp
#include <Wire.h>
#include <I2C_eeprom.h>

uint32_t eepromSize;
uint8_t b, c, d;

#define SERIAL_OUT Serial

I2C_eeprom ee(0x50, 4096);  //  <<<<<<<<<<<<<<< explicit size

void setup(){
  SERIAL_OUT.begin(9600);
  ee.begin();
  if(!ee.isConnected()){
    SERIAL_OUT.println("No memory detected. Freezing.");
    while(1);
  }
  SERIAL_OUT.println("Memory detected!");
  delay(10);
  SERIAL_OUT.print("Memory size in bytes: ");
  eepromSize = ee.determineSize();
  SERIAL_OUT.print(eepromSize);
  SERIAL_OUT.print('\n');

  SERIAL_OUT.println("\nWriteByte returns:");
  SERIAL_OUT.println(ee.writeByte(0x0000,0xA1), HEX);
  SERIAL_OUT.println(ee.writeByte(0x0001,0xB2), HEX);
  SERIAL_OUT.println(ee.writeByte(0x0002,0xC3), HEX);

  SERIAL_OUT.print("\nYour written data is: 0h");
  SERIAL_OUT.print(ee.readByte(0x0000),HEX);
  SERIAL_OUT.print(ee.readByte(0x0001),HEX);
  SERIAL_OUT.print(ee.readByte(0x0002),HEX);
  SERIAL_OUT.print('\n');
}

void loop(){
}
```

4r53N1ck, 2023-01-12:
Yes! Now it works. But why is that "determineSize()" worked before?

RobTillaart, 2023-01-12:
> Yes! Now it works. But why is that "determineSize()" worked before?

Yes, was pretty trivial in the end, I am used to using the default 32 KB EEPROMs, I hardly ever use others, so I missed it too.

DetermineSize() goes over all possible sizes and overrules temporary size settings. 
That might the reason it worked with the incorrect pageSize. 

Think I should add **void setDeviceSize(size)** and **setPageSize(size)** for debugging / overruling. 
Maybe change the example sketches to use explicit device sizes.

Note: 
You can use the constant: **I2C_DEVICESIZE_24LC32** that looks a bit like your EEPROM type,
and keep in mind that the write cycle can be much longer than the lib supports. up to 20 ms.

If there are no further questions you may close the issue (as I consider it solved)

4r53N1ck, 2023-01-12:
Many thanks to you! Best regards. Solved!

closed 2023-01-12 by RobTillaart