RobTillaart/I2C_EEPROM #53: Using the library in RP2040

jotamachuca, 2023-05-02:
Hi !

I used your library with the RP2040 (Raspberry Pi Pico) . I used the earlephilhower board for my project. 

For used your library in my project I make this changes in the I2c_eeprom.h and  I2C_eeprom.cpp file

.h 

```
#if defined (ESP8266) || defined(ESP32) ||defined(PICO_RP2040)
  //  set the I2C pins explicitly (overrule)
  bool     begin(uint8_t sda, uint8_t scl);
#endif
```

```
#ifdef PICO_RP2040
bool I2C_eeprom::begin(uint8_t sda, uint8_t scl)
{
   //  if (_wire == 0) Serial.println("zero");  //  test #48
  if ((sda < 255) && (scl < 255))
  {
    _wire->setSCL( scl );
    _wire->setSDA( sda);
    _wire->begin();
  }
  _lastWrite = 0;
  return isConnected();
}
#endif 
```

I want to add this change in the actual branch. Can it ??

Thanks for your work !!

See you later

RobTillaart, 2023-05-02:
Thanks for this addition,

You can make a PR, or have a little patience and I will do it later today.

RobTillaart, 2023-05-02:
I propose I will do it as I have might have some minor edits I can include.
Expect a develop branch in 30 minutes.

jotamachuca, 2023-05-02:
Nice ! 

I will wait and add this little addition :D

RobTillaart, 2023-05-02:
FYI,
https://github.com/RobTillaart/I2C_EEPROM/pull/54  Pull Request created, build is running.

RobTillaart, 2023-05-02:
@jotamachuca 
Build ran successfully, will merge it after lunch 
You may download the develop branch and test it if you are in a hurry :)

RobTillaart, 2023-05-02:
Did some minor edits,

RobTillaart, 2023-05-02:
Merged develop branch into master, releasing 1.7.3
Thanks for the addition.

RobTillaart, 2023-05-02:
@jotamachuca 

I was porting your proposal for improved RP2040 support to the I2C_24LC1025 library and I noticed another possible improvement for the RP2040. Don't know if you need it.

I looked it up and the internal I2C buffer size of the RP2040 ==> 256 bytes

I2C_eeprom.cpp line 25-30
```cpp
#if defined(ESP32) || defined(ESP8266)
#define I2C_BUFFERSIZE           128
#else
#define I2C_BUFFERSIZE           30   //  AVR, STM
#endif
```

==>

```cpp
#if defined(ESP32) || defined(ESP8266) || defined(PICO_RP2040)
#define I2C_BUFFERSIZE           128
#else
#define I2C_BUFFERSIZE           30   //  AVR, STM
#endif
```

For EEPROM's the page size is at most 64 KB (afaik) so using bigger buffers will not increase performance (assumption).

Do you have time to see if it gives a performance gain?

I make a note of it so it will be in a next release (someday).

RobTillaart, 2023-05-02:
(added the code in my local branch so it will automatically in the next release)

jotamachuca, 2023-05-02:
I will check this mods this week :D

jotamachuca, 2023-05-07:
Hi !!

I checked and all is ok :D

RobTillaart, 2023-05-07:
Good to hear!

RobTillaart, 2023-05-09:
@jotamachuca 

Apparently your **defined(PICO_RP2040)** is not unique enough to be **earlephilhower** specific.
See #55

jotamachuca, 2023-05-09:
> @jotamachuca
> 
> Apparently your **defined(PICO_RP2040)** is not unique enough to be **earlephilhower** specific. See #55

Uff

I will check tomorrow what is the correct defined for the earlephilhower .... 

Sorry for this mistake, but, for me, the library work without problems. 

See you tomorrow!

RobTillaart, 2023-05-09:
Thanks for the quick response!

RobTillaart, 2023-05-10:
Created a develop branch with a patch that should repair the backwards compatibility.

You need to uncomment line 33 in I2C_EEPROM.h  for now.
I have send a mail to earlephilhower to ask if his boards module has an unique define to fix this issue.

Please verify.

jotamachuca, 2023-05-10:
Hi

I checked the code very fast and,  I cannot found a define for earlephilhower core .....

I will write a email to check it 

Sorry for it problem

jotamachuca, 2023-05-10:
> The line (33) you should uncomment in the develop branch is
> 
> I2C_EEEPROM.h
> 
> ```c++
> #define RP2040_PATCH_53             true
> ```

Sure, but i want to find the define for the earlephilhower core.

RobTillaart, 2023-05-10:
> Sure, but i want to find the define for the earlephilhower core.

Searched his code and found many defines but none of them unique as far as I could see,
Therefor I decided to email him this morning, if such define exists.
In the mean time I patched with a pretty unique define to prevent breaking too much projects.

jotamachuca, 2023-05-10:
> > Sure, but i want to find the define for the earlephilhower core.
> 
> Searched his code and found many defines but none of them unique as far as I could see, Therefor I decided to email him this morning, if such define exists. In the mean time I patched with a pretty unique define to prevent breaking too much projects.
> 
> But if you find something, please let me know.

Yes, i checked today in the morning, and i havent an unique define for used.... 

I will wait for the answer :D 

See you later !!

RobTillaart, 2023-05-10:
@jotamachuca 

Got an answer and **ARDUINO_ARCH_RP2040** was proposed.
As that is also used by the Arduino MBED , it is not unique enough, so investigation continues....

RobTillaart, 2023-05-10:
An option is to combine **ARDUINO_ARCH_RP2040** and **ARDUINO_ARCH_MBED**.

```
#if defined(ARDUINO_ARCH_RP2040)
###
#endif
#if defined(ARDUINO_ARCH_MBED)
####
#endif

RobTillaart, 2023-05-11:
Merged and released 1.7.3 with better solution.

closed 2023-05-11 by RobTillaart

jotamachuca, 2023-05-11:
I used the  ARDUINO_ARCH_RP2040 and all ok :D

commit c4400ea1 (2023-05-02): fix #53 PICO_RP2040 support (#54)

- fix #53 support RP2040 (kudos to jotamachuca)
- move code from .h to .cpp
- make I2C_WRITEDELAY overridable
- minor edits
commit a13ec203 (2023-05-11): fix #55, redo #53 (#56)

- fix #55 ==> redo fix #53
  - add test to detect **MBED** and **RP2040**
- adjust **I2C_BUFFERSIZE** for RP2040 to 128.
- update readme.md