RobTillaart/I2C_EEPROM #55: Compile error for Arduino rp2040 linked to _wire->setSCL(scl) and _wire->setSDA(sda)

SebKister, 2023-05-09:
Hi ! 

After updating to v1.7.2 I am getting the following error when compiling my project for arduino rp20240:

Compiling .pio\build\nanorp2040connect\lib376\I2C_EEPROM\I2C_eeprom.cpp.o
.pio\libdeps\nanorp2040connect\I2C_EEPROM\I2C_eeprom.cpp: In member function 'bool I2C_eeprom::begin(uint8_t, uint8_t)':
.pio\libdeps\nanorp2040connect\I2C_EEPROM\I2C_eeprom.cpp:79:12: error: 'TwoWire' {aka 'class arduino::MbedI2C'} has no member named 'setSCL'
   79 |     _wire->setSCL(scl);
      |            ^~~~~~
.pio\libdeps\nanorp2040connect\I2C_EEPROM\I2C_eeprom.cpp:80:12: error: 'TwoWire' {aka 'class arduino::MbedI2C'} has no member named 'setSDA'
   80 |     _wire->setSDA(sda);

This was not the case with v1.7.1

Thanks !

RobTillaart, 2023-05-09:
Thanks for this issue,
It is rather late now her but I will look into this tomorrow!

It is indeed introduced in the last release after - https://github.com/RobTillaart/I2C_EEPROM/issues/53

Apparently there are differences between development platforms as **defined(PICO_RP2040)** is not selective enough.
Which platform are you using?

RobTillaart, 2023-05-09:
Note to myself:
boards.txt ==> pico.compiler.mbed.arch.define=-DARDUINO_ARCH_RP2040

Can you try this quick patch?

I2C_EEPROM.h   change line 62++
```cpp
#if defined(ARDUINO_ARCH_RP2040)
#elif defined(ESP8266) || defined(ESP32) || (defined(PICO_RP2040)
  //  set the I2C pins explicitly (overrule)
  bool     begin(uint8_t sda, uint8_t scl);
#endif
```

I2C_EEPROM.cpp   change line 56++
```cpp

#if defined(ARDUINO_ARCH_RP2040)
#elif defined(ESP8266) || defined(ESP32) || defined(PICO_RP2040)
bool I2C_eeprom::begin(uint8_t sda, uint8_t scl)
{
   //  if (_wire == 0) Serial.println("zero");  //  test #48
  if ((sda < 255) && (scl < 255))
  {
    _wire->begin(sda, scl);
  }
  else
  {
    _wire->begin();
  }
  _lastWrite = 0;
  return isConnected();
}
#endif
```

To be continued.

RobTillaart, 2023-05-09:
could do a quick test and the above is no solution.

To continue tomorrow, it is getting too late ..

SebKister, 2023-05-09:
Thanks Rob for the quick response ! That did the trick on my side.
This is the platformio.ini content of my project:

[env:nanorp2040connect]
platform = raspberrypi
board = nanorp2040connect
framework = arduino
upload_port = COM21
lib_ldf_mode = deep
build_flags = -DARDUINO_NANO_RP2040_CONNECT
...

so indeed ARDUINO_ARCH_RP2040 is defined

RobTillaart, 2023-05-10:
Send a mail to earlephilhower to ask if his boards module has an unique define to fix this issue.

RobTillaart, 2023-05-10:
@SebKister 
Created a develop branch with a patch that should make it work again.
Please verify.

SebKister, 2023-05-10:
Hi  Rob,

develop branch fixes issues on my side. Thanks !

RobTillaart, 2023-05-10:
@SebKister 
Good to hear,

I might have found a solution by combining **ARDUINO_ARCH_RP2040** and **ARDUINO_ARCH_MBED**.

```cpp
#if defined(ARDUINO_ARCH_RP2040) 
#if not defined(ARDUINO_ARCH_MBED)
  begin(sda, scl)     //  earlephilhower 
#endif
#endif 
```

to be continued tomorrow.

commit a13ec203 (2023-05-11): fix #55, redo #53 (#56)

- fix #55 ==> redo fix #53
  - add test to detect **MBED** and **RP2040**
- adjust **I2C_BUFFERSIZE** for RP2040 to 128.
- update readme.md
closed 2023-05-11 by RobTillaart