RobTillaart/I2C_EEPROM #57: Support for WP-Pins

F-Schmidt99, 2023-09-06:
Could you please add support for Write-Protect-Pins? I always connect them to a GPIO of my microcontroller which allows for better control of the EEPROM, but sadly as far as I can see, your library does not support this feature.

It would be awesome to see that addition =)

RobTillaart, 2023-09-06:
Thanks for the request, 
Have thought about it many times but never implemented it as it is quite simple to implement for the user.

Have you thoughts how the interface should look?

my first thoughts would be 
```cpp
void setWriteProtectPin(uint8_t pin, bool enable);
void disableWrite();
void enableWrite();
```

RobTillaart, 2023-09-06:
2nd thought is to follow my FRAM library

```cpp
  int      begin(....,  const int8_t writeProtectPin = -1);
  //  WRITEPROTECT
  //  works only if pin is defined in begin().
  bool     setWriteProtect(bool b);
  bool     getWriteProtect();
```

F-Schmidt99, 2023-09-06:
I'd recommend something even more simple to use:

Passing the pin to a begin function seems nice. One could also pass the pin number to the constructor. Use whatever you prefer ;)

But for the usage: Just toggle the pin in every update/write function. This allows the user to use the currently implemented methods the same way as before. You know, that WP has to be disabled on write - and ONLY on write and as short as possible, so there is no need to implement a setWriteProtect method or similar.

F-Schmidt99, 2023-09-06:
This is an implementation that I currently use and it works like a charm:

```
class Ulm_EEPROM_I2C_24LC16_WP : public I2C_eeprom {

private:
    uint8_t pinWP;

public:
    explicit Ulm_EEPROM_I2C_24LC16_WP(int8_t pinWP = -1);

    bool begin();
    bool updateByteVerify(uint16_t memoryAddress, uint8_t value) override;

private:
    void disableWriteProtect() const;
    void enableWriteProtect() const;
};
```
and the CPP:

```
Ulm_EEPROM_I2C_24LC16_WP::Ulm_EEPROM_I2C_24LC16_WP(int8_t pinWP)
        : I2C_eeprom(0x50, I2C_DEVICESIZE_24LC16, &Wire) {
    this->pinWP = pinWP;
}

bool Ulm_EEPROM_I2C_24LC16_WP::begin() {
    if(this->pinWP >= 0) {
        pinMode(this->pinWP, OUTPUT);
        digitalWrite(this->pinWP, HIGH);
    }
    return I2C_eeprom::begin();
}

bool Ulm_EEPROM_I2C_24LC16_WP::updateByteVerify(const uint16_t memoryAddress, const uint8_t value) {
    this->disableWriteProtect();
    bool result = I2C_eeprom::updateByteVerify(memoryAddress, value);
    this->enableWriteProtect();
    return result;
}

void Ulm_EEPROM_I2C_24LC16_WP::disableWriteProtect() const {
    if(this->pinWP >= 0) {
        digitalWrite(this->pinWP, LOW);
        delayMicroseconds(20);
    }
}

void Ulm_EEPROM_I2C_24LC16_WP::enableWriteProtect() const {
    if(this->pinWP >= 0) {
        digitalWrite(this->pinWP, HIGH);
        delayMicroseconds(20);
    }
}
```

Please note, that I have only added the WP-mechanism to the updateByteVerify method. Of course, this must apply to all writing functions!

RobTillaart, 2023-09-06:
Why do you want to set the WP pin?
If the application is allowed to write, you connect WP to 5V and if you want the EEPROM in read only mode connect it to GND (Oir vice versa, datasheet)

What is the rationale to automatically control the WP pin when writing / updating? 

The only reason I can think of to have the WP under application control is that only certain people /processes or even time slots are allowed to write.
Then you want to enable the WP until user is done.

Enabling /disabling every write / update makes it a tiny bit slower (could be acceptable) however I do not see the advantage.

F-Schmidt99, 2023-09-06:
I am using the circuit in very rough environments and want to make sure that there are no external sources of interference that might change data on my bus which results in a write. And therefore I toggle the WP pin only when needed, so directly before and after writing to it.

Slower speed is true (you need to wait 2x toggle time more), but I do not mind that at all, because only very few bytes are written. For my application it is more beneficial to make sure, that WP is active.

RobTillaart, 2023-09-06:
OK, makes more sense now
 (thinking out loud)
What you propose is an automatic mode,
While I am thinking of a manual mide.

Need to do some impact analysis.
(Loose thoughts)
- extend class, 2 bytes RAM, a dozen extra lines stc
- derived class is option
- conditional define could enable it (keep footprint small for those not using WP.

Will come back to it later this week.

RobTillaart, 2023-09-06:
Found some time to work on your request.
Created a develop branch and a Pull Request - https://github.com/RobTillaart/I2C_EEPROM/pull/58
(changed some minor things too).

The readme.md file has a small section about the usage of the new functions.
- you must define the WP pin in **begin()**
- in your use case you must set **setAutoWriteProtect(true)** in your setup.

Can you give it a try?
Labeled it as experimental for now as I had no time to do a test with hardware myself.

Furthermore I have not written an example yet, 
to be continued!

RobTillaart, 2023-09-08:
@F-Schmidt99 
prepared the develop branch for merge. 
Is the readme clear enough in your opinion (WP part).

https://github.com/RobTillaart/I2C_EEPROM/blob/develop/README.md#writeprotectpin-wp-experimental

F-Schmidt99, 2023-09-08:
The only question that arises is, if it is enough to have this code:

```
if(eeprom.begin()) {
    eeprom.setAutoWriteProtect(true);
}
```

Does this code lock the chip right away?

RobTillaart, 2023-09-08:
No, the code should be

```cpp
#define WP_PIN 42  // whatever

if(eeprom.begin(WP_PIN)) {
    eeprom.setAutoWriteProtect(true);
}
```

RobTillaart, 2023-09-08:
Then I will merge right now.

closed 2023-09-08 by RobTillaart

F-Schmidt99, 2023-09-08:
Awesome!

RobTillaart, 2023-09-08:
builds are running, (will be picked up later by platformIO / library manager)

F-Schmidt99, 2023-09-08:
yep, I will check this afternoon. Thank you for the fast update =)