RobTillaart/I2C_EEPROM #64: Compiler warnings

thijstriemstra, 2024-03-21:
Using platformio:

```ini
[env:pico]
platform = raspberrypi
board = pico
framework = arduino

lib_deps =
  robtillaart/I2C_EEPROM@^1.8.2
```

seeing some warnings while compiling my project:

```
.pio\libdeps\pico\I2C_EEPROM\I2C_eeprom.cpp: In member function 'bool I2C_eeprom::writeBlockVerify(uint16_t, const uint8_t*, uint16_t)':
.pio\libdeps\pico\I2C_EEPROM\I2C_eeprom.cpp:206:11: warning: variable length array 'data' is used [-Wvla]
  206 |   uint8_t data[length];
      |           ^~~~
.pio\libdeps\pico\I2C_EEPROM\I2C_eeprom.cpp: In member function 'bool I2C_eeprom::setBlockVerify(uint16_t, uint8_t, uint16_t)':
.pio\libdeps\pico\I2C_EEPROM\I2C_eeprom.cpp:216:11: warning: variable length array 'data' is used [-Wvla]
  216 |   uint8_t data[length];
      |           ^~~~
.pio\libdeps\pico\I2C_EEPROM\I2C_eeprom.cpp: In member function 'bool I2C_eeprom::updateBlockVerify(uint16_t, const uint8_t*, uint16_t)':
.pio\libdeps\pico\I2C_EEPROM\I2C_eeprom.cpp:239:11: warning: variable length array 'data' is used [-Wvla]
  239 |   uint8_t data[length];
      |           ^~~~

RobTillaart, 2024-03-21:
#### Analysis
data[length] is used as a buffer to verify / compare.

#### Possible solution
Maybe a **_verifyBlock(addr, data, length)** that verifies while reading. Could work per byte, uses less memory and fails faster?

RobTillaart, 2024-03-23:
@thijstriemstra 
Created a develop branch for testing 
- added **verifyBlock()** for readBlockVerify() and writeBlockVerify 
- that did not fit the setBlockVerify so I solver that with dynamic memory for now. 

Can you check if the warnings still exist (or are replaced with others)?

thijstriemstra, 2024-03-23:
> Can you check if the warnings still exist (or are replaced with others)?

I'll give it a try. Might be a good idea to setup a platformio builder and make it fail on warnings like this.

RobTillaart, 2024-03-23:
That is a good idea, 
Do you know if there is a GitHub workflow that I could use?
Then I could roll it out quite easily for all libraries.

thijstriemstra, 2024-03-23:
See https://docs.platformio.org/en/latest/integration/ci/github-actions.html

so something like:

```
name: PlatformIO CI

  on: [push]

  jobs:
    build:
      runs-on: ubuntu-latest
      strategy:
        matrix:
          example: [path/to/test/file.c, examples/file.ino, path/to/test/directory]

      steps:
        - uses: actions/checkout@v4
        - uses: actions/cache@v3
          with:
            path: |
              ~/.cache/pip
              ~/.platformio/.cache
            key: ${{ runner.os }}-pio
        - uses: actions/setup-python@v5
          with:
            python-version: '3.11'
        - name: Install PlatformIO Core
          run: pip install --upgrade platformio

        - name: Build PlatformIO examples
          run: pio ci --board=<ID_1> --board=<ID_2> --board=<ID_N>
          env:
            PLATFORMIO_CI_SRC: ${{ matrix.example }}
```

thijstriemstra, 2024-03-23:
And to make it fail on warnings add `-Werror` (see https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html):

> Make all warnings into hard errors. With this option, if any source code triggers warnings, the compilation will be aborted.

So:

```
- name: Run PlatformIO
  run: pio ci path/to/test/file.c --board=<ID_1> --board=<ID_2> --board=<ID_N>
  env:
    PLATFORMIO_BUILD_FLAGS: -D -Werror
```

(untested)

RobTillaart, 2024-03-23:
Created an issue for this in a smaller library to investigate and get it to work
- https://github.com/RobTillaart/A1301/issues/7

thijstriemstra, 2024-03-23:
Cool! The only disadvantage i can see is compiler warnings in third party libraries/frameworks, that'll fail the build as well. Narrowing warnings down to lib only code might also be possible but not sure/tested.

RobTillaart, 2024-03-23:
OK, I will try to find out coming days/weeks (quite busy with other projects) how to get the platformIO build running.

RobTillaart, 2024-03-23:
Got something building with green results 😎

- https://github.com/RobTillaart/A1301/actions/runs/8403081050/job/23013196189

yaml file is not portable or engineered (yet) but there is a starting point.

RobTillaart, 2024-03-23:
The -Werror fails upon the Arduino core libs like HWSerial, so that switch will not be used...

```
Building in release mode
Compiling .pio/build/uno/src/A1301_autoMidPoint.ino.cpp.o
Compiling .pio/build/uno/lib954/A1301/A1301.cpp.o
Archiving .pio/build/uno/libFrameworkArduinoVariant.a
<command-line>:0:1: error: macro names must be identifiers
Compiling .pio/build/uno/FrameworkArduino/CDC.cpp.o
<command-line>:0:1: error: macro names must be identifiers
Indexing .pio/build/uno/libFrameworkArduinoVariant.a
<command-line>:0:1: error: macro names must be identifiers
Compiling .pio/build/uno/FrameworkArduino/HardwareSerial.cpp.o
<command-line>:0:1: error: macro names must be identifiers
*** [.pio/build/uno/src/A1301_autoMidPoint.ino.cpp.o] Error 1
*** [.pio/build/uno/lib954/A1301/A1301.cpp.o] Error 1
*** [.pio/build/uno/FrameworkArduino/CDC.cpp.o] Error 1
*** [.pio/build/uno/FrameworkArduino/HardwareSerial.cpp.o] Error 1
========================== [FAILED] Took 0.[65](https://github.com/RobTillaart/A1301/actions/runs/8403259599/job/23013574092#step:7:66) seconds ==========================
```

RobTillaart, 2024-03-23:
compiling all examples by coding a hard list works
- https://github.com/RobTillaart/A1301/actions/runs/8403378082

to be continued later.

RobTillaart, 2024-03-24:
Notes to me:
current test code A1301 repo

```yml

name: PlatformIO CI

on: [push, pull_request]

jobs:
  build:
    runs-on: ${{ matrix.os }}
    #  added timeout
    timeout-minutes: 20
    strategy:
      matrix:
        os: [ubuntu-latest]
        example: [
                   examples/A1301_autoMidPoint/A1301_autoMidPoint.ino,
                   examples/A1301_demo/A1301_demo.ino,
                   examples/A1301_determineNoise/A1301_determineNoise.ino,
                   examples/A1301_performance/A1301_performance.ino,
                   examples/A1301_plotter/A1301_plotter.ino,
                   examples/A1301_test/A1301_test.ino,
                   examples/A1301_test_saturation/A1301_test_saturation.ino
                  ]
 
    steps:
      - uses: actions/checkout@v4
      - uses: actions/cache@v3
        with:
          path: |
            ~/.cache/pip
            ~/.platformio/.cache
          key: ${{ runner.os }}-pio
      - uses: actions/setup-python@v5
        with:
          python-version: '3.11'
      - name: Install PlatformIO Core
        run: pip install --upgrade platformio
      - name: Download library
        run: |
          wget https://github.com/RobTillaart/A1301/archive/master.zip -O /tmp/master.zip
          unzip /tmp/master.zip -d /tmp
      - name: Build PlatformIO examples
        run: pio ci --lib "." --lib="/tmp/A1301-master" --board=uno
        env:
          PLATFORMIO_CI_SRC: ${{ matrix.example }}
          #  PLATFORMIO_BUILD_FLAGS: -D -Wall  # too much errors in core Arduino.
```
hardcoded are
- examples to compile
- repository to checkout
- branch to checkout

These should ideally be created / derived from the branch pushed.
Some options to get it done:
- maintain manually
- python script generating the yml file
- python script that runs on VM and do the matrix
- do magic in yml file

RobTillaart, 2024-03-24:
After reading 
- https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions

I could fix the repository and correct branch (first it was hardcoded master while it had to be a develop branch).

```yml
      - name: Download library
        run: |
          wget https://github.com/${{github.repository}}/archive/${{github.ref_name}}.zip -O /tmp/${{github.ref_name}}.zip
          unzip /tmp/${{github.ref_name}}.zip -d /tmp/${{github.actor}}
      - name: Build PlatformIO examples
        run: pio ci --lib "." --lib="/tmp/${{github.repository}}-${{github.ref_name}}" --board=uno
        env:
          PLATFORMIO_CI_SRC: ${{ matrix.example }}
          #  PLATFORMIO_BUILD_FLAGS: -D -Wall  # too much 
```

Only the list of examples files need to be generated,.
Maintaining that list manually is doable as the list does not change often.

RobTillaart, 2024-03-25:
Did a test with ESP32DEV instead of UNO, also warnings in the ESP32 specific files.
So I doubt if the **Wall** option will work 

```
Compiling .pio/build/esp32dev/FrameworkArduino/libb64/cdecode.c.o
/home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c: In function 'uartSetPins':
/home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void
         return;
         ^~~~~~
```

This bug is already reported at - https://github.com/espressif/arduino-esp32/issues/8641

RobTillaart, 2024-03-27:
@thijstriemstra 

> I'll give it a try. Might be a good idea to setup a platformio builder and make it fail on warnings like this.

Did you have time to try the develop branch?

thijstriemstra, 2024-03-27:
After switching to:

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipico
framework = arduino
monitor_speed = 115200
; board can use both Arduino cores -- we select Arduino-Pico here
board_build.core = earlephilhower
; Flash Size: 2MB (Sketch: 1.5MB, FS: 0.5MB)
board_build.filesystem_size = 0.5m
```

the compile errors seem to be gone, and since I'll be using `earlephilhower` (since platformio raspberrypi is not supported by platformio labs and basically abandoned) I'll leave it up to you to implement your suggested fix or not.

RobTillaart, 2024-03-28:
OK, I will merge the develop branch today
- adds **verifyBlock(memoryAddress, buffer, length)**
- updates build-CI scripts

commit ad545109 (2024-03-23): fix #64, warning in compile
commit f3aae22d (2024-03-23): fix #64, warning in compile
commit f3e70ff3 (2024-03-28): implement verifyBlock (#67)

- Fix #64, compiler warning.
- add **verifyBlock(memoryAddress, buffer, length)**
- add example **I2C_eeprom_verifyBlock.ino**
- update GitHub actions
- update keywords.txt
- update examples
- update readme.md
closed 2024-03-28 by RobTillaart