RobTillaart/I2C_EEPROM #66: Help with I2C_eeprom_cyclic_store

thijstriemstra, 2024-03-27:
I'd like to use the I2C_eeprom_cyclic_store example as a template for a project I need to port:

```c++
// default values at startup, if nothing is stored in the eeprom
int DryTemperature = 35;    // Destination Dry Temperature in degree celsius
int DryTime_Hours = 1;
int DryTime_Minutes = 30;
int curDryTime_Hours = 1;
int curDryTime_Minutes = 30;

I2C_eeprom ee(EEPROM_ADDRESS, EEPROM_SIZE);

/***
 * Saves the current settings for temperature, drying hours and minutes
 */
void SaveSettings() {
  EEPROM.write(0, (uint8_t) 17);
  EEPROM.write(1, (uint8_t) DryTemperature);
  EEPROM.write(2, (uint8_t) DryTime_Hours);
  EEPROM.write(3, (uint8_t) DryTime_Minutes);
}

/***
 * Reads the last saved settings for temperature, drying hours and minutes
 */
void ReadSettings() {
  uint8_t isInit = EEPROM.read(0);

  if (isInit == 17) {
    DryTemperature = EEPROM.read(1);
    DryTime_Hours = EEPROM.read(2);
    DryTime_Minutes = EEPROM.read(3);
  }
}
```

but it's hard to figure out how to use that I2C_eeprom_cyclic_store example in a way that matches the old EEPROM library.

These lines for example:

```
#define MEMORY_SIZE 0x2000 // Total capacity of the EEPROM
#define PAGE_SIZE 64 // Size of write page of device, use datasheet to find!
```

I have no idea what to use here for my AT24C64D device on 0x50.

It feels a lot of the examples are more unit test type of examples and I'm missing simple hello world kind of stuff.

Any suggestions are welcome!

thijstriemstra, 2024-03-28:
I found a table in the readme (`Test results determineSize()`), indicating the page size for my `24LC64` is 32:

| Type    | returns |  Memory  | Page Size | Notes |
|:--------|:--------|:---------|:---------:|:------|
|  -      |    0    |          |           | connect error, check device address / wiring |
| 24LC512 |  65536  |  64 KB   |   128     |       |
| 24LC256 |  32768  |  32 KB   |    64     |       |
| 24LC128 |  16384  |  16 KB   |    64     |       |
| 24LC64  |   8192  |   8 KB   |    **32**     |       |
| 24LC32  |   4096  |   4 KB   |    32     | not tested with hardware |
| 24LC16  |   2048  |   2 KB   |    16     |       |
| 24LC08  |   1024  |   1 KB   |    16     |       |
| 24LC04  |    512  |  512 b   |    16     |       |
| 24LC02  |    256  |  256 b   |     8     |       |
| 24LC01  |    128  |  128 b   |     8     |       |

so I defined:

```c++
#define PAGE_SIZE 32
```

the default `MEMORY_SIZE` is `0x2000 `:

```
#define MEMORY_SIZE 0x2000 // Total capacity of the EEPROM
```

which, according to https://www.rapidtables.com/convert/number/hex-to-decimal.html?x=0x2000, equates to 8192, the correct size (when looking at table).

RobTillaart, 2024-03-28:
> It feels a lot of the examples are more unit test type of examples and I'm missing simple hello world kind of stuff.

Most examples come indeed from sketches testing the library, and yes, tutorials are not included. 
Also not all platforms/boards are supported and / or tested. Time is a scarce resource..

RobTillaart, 2024-03-28:
you could also used
```cpp
#define MEMORY_SIZE 8192  // Total capacity of the EEPROM
```
would work as it is just another representation of the same number.

RobTillaart, 2024-03-28:
Yes, a lot of documentation can be improved. I'll add a note in my local copy so it will come with the next update.

RobTillaart, 2024-04-04:
@thijstriemstra 

If the question seems to be solved you may close the issue.

closed 2024-04-04 by thijstriemstra