RobTillaart/I2C_EEPROM #68: Compatible code for M24512

killerbeat12, 2024-04-01:
Hi, could i use this library also for M24512?

RobTillaart, 2024-04-01:
Thanks for your question.
I am not familiar with that type number. Most EEPROM's I know work pretty similar. To be sure you need to compare the datasheet.

Do you have a link to its datasheet?

killerbeat12, 2024-04-01:
Thanks for your quick response, yes : 
[m24512_r-1849742.pdf](https://github.com/RobTillaart/I2C_EEPROM/files/14823907/m24512_r-1849742.pdf)

RobTillaart, 2024-04-02:
From the **M24512** datasheet -- compared to supported 24AA512/24LC512/24FC512 datasheet

> The M24512-D offers an additional page, named the identification page (128 byte). The identification page can be
used to store sensitive application parameters which can be (later) permanently locked in read-only mode.

[image]

[image]

[image]

This is not supported in my library, 

#### Pins

Pins look very similar
Write Control == Write protect  (works same)
E0, E1, E2 === A0, A1, A2 (works same)

#### Address

Address range is 0x50..0x57 is same for core EEPROM memory

The M24512-D also offers address 0x58..0x5F for the identification page.
==> last 3  enable/address bits same as core EEPROM.

#### Data transfer

Looks the same to me.

#### Conclusion

After a quick compare of the datasheets I think the library should work with the M24512 device.
At least the core EEPROM function of the M24512.

The only part that is not supported is the identification page of the M24512-D model.
This could in theory be supported in a derived class. 

So give it a try and let me know if your M24512 works.

RobTillaart, 2024-04-02:
@killerbeat12 
M24xxx series mentioned some years ago in #21 and one model was confirmed working.

RobTillaart, 2024-04-04:
@killerbeat12 
Any progress to report?

killerbeat12, 2024-04-04:
Hi Rob thanks for your email , no i am sorry i am a couple of days off and at this moment they are manufacturing the print , i guess after 2 weeks i will come back to you , thanks till so far and have a great day , Richard Verzonden vanaf Outlook voor AndroidVan: Rob Tillaart ***@***.***>Verstuurd: donderdag, april 4, 2024 3:03:17 p.m.Aan: RobTillaart/I2C_EEPROM ***@***.***>CC: killerbeat12 ***@***.***>; Mention ***@***.***>Onderwerp: Re: [RobTillaart/I2C_EEPROM] Compatible code for M24512 (Issue #68)
@killerbeat12
Any progress to report?

killerbeat12, 2024-04-04:
Ps , i just add you as connection on LinkedIn ( ik ben ook Nederlands ) are.you open for paid project support ? What is your our rate ? I live nearby , Oss Verzonden vanaf Outlook voor AndroidVan: info ***@***.***>Verstuurd: donderdag, april 4, 2024 3:14:02 p.m.Aan: RobTillaart/I2C_EEPROM ***@***.***>; RobTillaart/I2C_EEPROM ***@***.***>CC: Mention ***@***.***>Onderwerp: Re: [RobTillaart/I2C_EEPROM] Compatible code for M24512 (Issue #68)Hi Rob thanks for your email , no i am sorry i am a couple of days off and at this moment they are manufacturing the print , i guess after 2 weeks i will come back to you , thanks till so far and have a great day , Richard Verzonden vanaf Outlook voor AndroidVan: Rob Tillaart ***@***.***>Verstuurd: donderdag, april 4, 2024 3:03:17 p.m.Aan: RobTillaart/I2C_EEPROM ***@***.***>CC: killerbeat12 ***@***.***>; Mention ***@***.***>Onderwerp: Re: [RobTillaart/I2C_EEPROM] Compatible code for M24512 (Issue #68)
@killerbeat12
Any progress to report?

RobTillaart, 2024-04-04:
Linked in accepted!

RobTillaart, 2024-04-15:
Confirmed working by email.

closed 2024-04-15 by RobTillaart