RobTillaart/I2C_EEPROM #72: Version [1.8.4] - 2024-04-20 Get Failure

ClaudiusVi, 2024-04-22:
\Arduino\libraries\Wire\src/Wire.h:125:13: note: candidate 2: 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t)'

RobTillaart, 2024-04-22:
Thanks for reporting these issues, will have a look at it today.

RobTillaart, 2024-04-22:
Cannot replicate,
Can you post a minimal sketch that shows the issue?

RobTillaart, 2024-04-22:
@ClaudiusVi 
OK, think I found the cause, and created a develop branch with the fix.

Can you verify the develop branch works for you?

ClaudiusVi, 2024-04-22:
@Rob
I've already made the changes in the ESP32 code. Initially, I wasn't sure where to look, so I checked your library code to find where the issue occurs. Since I'm currently writing libraries for the Raspberry Pi 5 with the new GPIO chip, it made the process straightforward.
Incorporating an example in the documentation like the following would clarify things:

const uint8_t EEPROM_ADDRESS = 0x50; 
I2C_eeprom EEPROM_24LC512(EEPROM_ADDRESS, I2C_DEVICESIZE_24LC512); 

Our program is quite extensive and includes LoRa, various sensors, and control modules, which is why the I2C addresses are declared as constants, or mistakenly as integers in some cases.

Thanks for your help,

Claudius

RobTillaart, 2024-04-22:
You're welcome,

The problem is not in the constructor **I2C_eeprom EEPROM_24LC512(EEPROM_ADDRESS, I2C_DEVICESIZE_24LC512);** or its parameters, so an additional example would not help or being educative.

The error is introduced by the last PR which allows the I2C buffer to be larger than 256 bytes, making low level calls with 16 bit variables instead of 8 bit. This causes an ambiguity in the library call **requestFrom()**. By casting the parameters of that function explicit to **int** this ambiguity is solved.

If you have time, please verify the develop branch,

commit 0de303ff (2024-04-22): Fix #72, parameter type requestFrom() (#73)

- Fix #72, force requestFrom parameters to use int type
closed 2024-04-22 by RobTillaart