RobTillaart/I2C_EEPROM #74: Potential memory leak in the implementation of I2C_eeprom::setBlockVerify()

cmichailidis, 2024-10-08:
**Description**: I believe there is a potential memory leak in the implementation of `I2C_eeprom::setBlockVerify()`. This is the relevant code snippet: 

```
bool I2C_eeprom::setBlockVerify(const uint16_t memoryAddress, const uint8_t value, const uint16_t length)
{
  if (setBlock(memoryAddress, value, length) != 0) return false;
  uint8_t * data = (uint8_t *) malloc(length);
  if (data == NULL) return false;
  if (readBlock(memoryAddress, data, length) != length) return false;
  for (uint16_t i = 0; i < length; i++)
  {
    if (data[i] != value)
    {
      free(data);
      return false;
    }
  }
  free(data);
  return true;
}
```

**Problem** From what I can tell, this function dynamically allocates some memory for the `data` buffer using `malloc()`. It then tests whether the allocation was successful or not by comparing the return value with `NULL`. However, if the subsequent call to `I2C_eeprom::readBlock()` fails for whatever reason, we exit prematurely from the scope of `setBlockVerify()` without calling `free(data)` first. 

My first thought was that `I2C_eeprom::readBlock()` takes ownership of the `data` buffer since it is passed in its arguments list and automatically de-allocates the memory, if necessary. However, after taking a quick look at its implementation I could not verify this assumption. 

Curiously enough, the subsequent `return` statements in the implementation of `setBlockVerify()` are preceded by a `free(data);` statement which implies that `setBlockVerify()` has ownership over the `data` buffer and is therefore responsible for allocating and de-allocating its memory. 

Is this a memory leak or is there  something else I am missing entirely? My background is primarily in C++ and I don't have to deal with those issues often thanks to smart-pointers.

**Possible Fix** Assuming this is indeed a memory leak, I believe the following modification would suffice: 

```
bool I2C_eeprom::setBlockVerify(const uint16_t memoryAddress, const uint8_t value, const uint16_t length)
{
  if (setBlock(memoryAddress, value, length) != 0) return false;

  uint8_t * data = (uint8_t *) malloc(length);
  if (data == NULL) return false;

  if (readBlock(memoryAddress, data, length) != length)
  {
    free(data);      //< Missing memory de-allocation step
    return false;
  }

  for (uint16_t i = 0; i < length; i++)
  {
    if (data[i] != value)
    {
      free(data);
      return false;
    }
  }

  free(data);
  return true;
}
```

RobTillaart, 2024-10-09:
Thanks for reporting,
This is definitely like a bug, 
Your proposed solution is correct.

Will work on new release asap.

RobTillaart, 2024-10-09:
Checked the remaining code and this is the only spot where dynamic allocation is used. 

Todo: verify - https://github.com/RobTillaart/I2C_24LC1025 - which is closely related.

RobTillaart, 2024-10-09:
PR is building.

RobTillaart, 2024-10-09:
Verified - https://github.com/RobTillaart/I2C_24LC1025 does not use malloc().

RobTillaart, 2024-10-09:
Merged PR and start release 1.9.0

Again thanks for catching this bug!

commit 48cd8e6e (2024-10-09): Fix #74, memory leak in verifyBlock()
commit 8cb180d0 (2024-10-09): Fix #74, memory leak in verifyBlock() (#75)

- Fix #74, memory leak in setBlockVerify()
closed 2024-10-09 by RobTillaart