RobTillaart/I2C_EEPROM #76: Use of Software-I2C ?

Greece2, 2024-10-22:
Any change of using this nice lib with software-I2C, like softwire, SWIIC, BitBang etc. ?
Would be a great addition ! :-)

Thanks, Greece2

RobTillaart, 2024-10-22:
If you have a soft-I2C that implements the TwoWire interface, all my i2c based libraries will work without any change.

So yes it can be done,

Greece2, 2024-10-22:
Hi Rob, that sounds great :-)
Though I'm afraid that you have to point me in the right direction ;-)
I used your example _I2C_eeprom_test_ and replaced the wire.h lib with the SoftwareWire.h lib (works fine in other projects, with i.e. the DS1307 RTC). 
Unfortunately, the changed code doesn't compile. 
The compiler complains that `I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC256, myWire);` isn't correct :-(

> W:\libraries\I2C_EEPROM/I2C_eeprom.h:80:3: note:   no known conversion for argument 3 from 'SoftwareWire' to 'TwoWire*'

Here's the modified code (only the first part...), may be you can tell where my error is (possibly something stupid ;-)  )?
```
//#include "Wire.h"
#include <SoftwareWire.h>
// SoftwareWire constructor.
// Parameters:
//   (1) pin for the software sda
//   (2) pin for the software scl
//   (3) use internal pullup resistors. Default true. Set to false to disable them.
//   (4) allow the Slave to stretch the clock pulse. Default true. Set to false for faster code.
// sda, scl, Pullups on, ClockStretching on
#define SW_SDA 49 // 5 // SDA
#define SW_SCL 48 //4 // SCL
SoftwareWire myWire(SW_SDA, SW_SCL, true, true);

#include "I2C_eeprom.h"

//  UNO
#define SERIAL_OUT Serial
//  Due
// #define SERIAL_OUT SerialUSB

/*
      @param deviceAddress Byte address of the device.
      @param deviceSize    Max size in bytes of the device (divide your device size in Kbits by 8)
      @param wire          Select alternative Wire interface

  I2C_eeprom(const uint8_t deviceAddress, const uint32_t deviceSize, TwoWire *wire = &Wire);
*/
// I2C_eeprom ee(0x50);
I2C_eeprom ee(0x50, I2C_DEVICESIZE_24LC256, myWire);

uint32_t start, diff, totals = 0;

void setup()
{
  SERIAL_OUT.begin(115200);
  while (!SERIAL_OUT);  //  wait for SERIAL_OUT port to connect. Needed for Leonardo only
  SERIAL_OUT.println(__FILE__);
  SERIAL_OUT.print("I2C_EEPROM_VERSION: ");
  SERIAL_OUT.println(I2C_EEPROM_VERSION);

  // Wire.begin();
  myWire.begin(); // start SW-I²C
  ee.begin();
```

Many thanks for your kind support !

Regards,  Greece2

RobTillaart, 2024-10-22:
You might find some clues in this triple of libraries where I investigated and implemented software I2C.

- https://github.com/RobTillaart/SHT31
- https://github.com/RobTillaart/SHT31_SW
- https://github.com/RobTillaart/SHT31_SWW

Took a long time to get these SW I2C variants working (sort of ).

Greece2, 2024-10-23:
Hey, great, that does help a lot. So I might have to modifiy the lib a little according to your SHT31...-examples .
Thanks a lot for the perfect support !  :-)

Regards, Greece2

RobTillaart, 2024-10-23:
Success!
Please keep me informed if it works!
(you may close the issue)

closed 2024-10-23 by RobTillaart