RobTillaart/I2C_EEPROM #82: Compatibility with 1Mb EEPROM chips?

dzalf, 2025-05-20:
Dear @RobTillaart 

It's me again, now in a different repo of yours 😉 

First of all, the library works flawlessly, however, during a development session to dump the entire contents of the EEPROM I realised that I had not set the size of my chip. This led me to ask myself the following question:

I would like to ask you if you have tested your library with 1Mb EEPROMs, like the [CAT24M01](https://www.onsemi.com/download/data-sheet/pdf/cat24m01-d.pdf) (or even bigger ones). I assume the library should work without any trouble.

I am aware that the library was designed for 512Mb chips maximum and that modifying it to support bigger sizes would require changing the values like the device size: `I2C_DEVICESIZE_24LC512 ` for `I2C_DEVICESIZE_24LC1024` (for instance), then page size would also change to 256: `I2C_PAGESIZE_24LC1024` and perhaps the `setPageSize` should include this new size:

```cpp
uint8_t I2C_eeprom::setPageSize(uint8_t pageSize)
{
  // force power of 2.

  if (pageSize >= 256) {
      _pageSize = 256;
  }
  if (pageSize >= 128) {
      _pageSize = 128;
  }
  else if (pageSize >= 64) {
      _pageSize = 64;
  }
  else if (pageSize >= 32) {
      _pageSize = 32;
  }
  else if (pageSize >= 16) {
      _pageSize = 16;
  }
  else {
      _pageSize = 8;
  }
  return _pageSize;
}
```

Similarly, `getPageSize` would change.

Are there any other changes I am not considering that would have breaking implications by modifying these values?

Would it be worth adding these changes?

Thanks

Best wishes

RobTillaart, 2025-05-20:
Welcome back,
Have other prio tasks the coming days so will be next week before I can dive into your question.
Improvements are always worth to investigate

RobTillaart, 2025-05-20:
Some notes
There is a bug in proposed code. 
It is missing an else.

To check Is  _pageSize 16 or 8 bit?
Can it hold 256?

@dzalf 
have you checked this one?
https://github.com/RobTillaart/I2C_24LC1025

RobTillaart, 2025-05-21:
Important to keep performance/ footprint if possible.
A derived class might be an option to consider?!

RobTillaart, 2025-05-25:
Still very occupied, so some thoughts  - no time yet to dig deep (sorry).

> I did check the mentioned library, however, in my case I have already designed part of my wrapper library based on this other library and I feel it will be a straightforward fix.

The interface of both libraries are very similar, the difference is mainly 16/32 bit.  So porting your wrapper should not be difficult.
I do not recall the original rationale why I made a separate library for this however I strongly assume the 32 bit would increase footprint and decrease performance for smaller boards with smaller EEPROMs.

A quick look at https://github.com/RobTillaart/I2C_24LC1025 shows it uses a 128-Byte Page Write Buffer.

As the CAT24M01 uses 32 bit addresses, the easiest library to patch should be the 24LC1025.
A second thought as the write buffer is rather "large" it might be wise to implement a separate library 
(need to investigate impact of the large write buffer on the footprint / local variables etc).

Minor diff noticed is that the 2 versus 3 address pins. 
See no issue for the library immediately as A2 must be hard wired for the LC1025.

RobTillaart, 2025-05-25:
> like the CAT24M01 (or even bigger ones).

Do you have links to the bigger ones? 
Smaller one seem to work with the I2C_EEPROM library.

dzalf, 2025-05-27:
Dear Rob,

Apologies for the delay in getting back to you — I’ve been quite busy, both with upgrading your library and handling other programming tasks.

I’ve developed a working version that extends your library’s functionality to support the 1 MB chip with 256-byte pages. The only minor issue I encountered was handling memory writes that span across different banks, due to changes in the a16 bit as specified in the datasheet.

I tested the updated library using a wrapper I wrote to perform a comprehensive showcase test, and everything appears to be functioning correctly.

I’ll need a bit more time to package a standalone version of the revised library for you to try. I believe the wrapper test code could serve well as an example sketch rather than a wrapper, in which case I’ll convert it into an independent example for clarity.

I’ll follow up with you later this week.

Thanks again for your time and support.

Best regards,

RobTillaart, 2025-05-27:
Is your work on github?
If so a link is appreciated.

dzalf, 2025-05-28:
Dear Rob

Apologies for the late reply.

I managed to "translate" my wrapper example into a standalone Arduino example code for your analysis.

I have created a new repo where I will put two or three example codes. The repo contains one example code (for now). Link: https://github.com/dzalf/I2C_EEPROM_1Mb_Devel

The modified library can be found here: https://github.com/dzalf/I2C_EEPROM_GT_Fork/tree/add-1024Mb (branch of interest is `add-1024Mb`).

Just so you know, I typically work on VSCode + platformio and so I was not sure what would be the best strategy to merge everything into a single folder while keeping the development paths reachable for the Arduino IDE. 

One important clarification is that I am developing code using two custom PCBs: one with the ESP32-S2 ( and based on the Lolin S2 Mini) and another system based on the ESP32-WROVER-IE since I needed a versatile way to debug the code for both systems. This is the reason why you will see a "weird" debugging initialization at the top of the code.

While the ESP32-S2 uses CDC USB for debugging, the WROVER-IE is connected to a CP2102N to be used with `HardwareSerial` via a selected UART.

The code `I2C_EEPROM_Comprehensive_Test.ino` (with the function `testGranularPatterns();` commented out) should show the following serial debugging output:

``` console

--- EEPROM Extended Test Suite ---
Reading default device size: 131072

--- Writing Settings to EEPROM ---
Writing data to EEPROM. Start Address: 0x0, Length: 32
Data successfully written and verified.

--- Reading Settings from EEPROM ---
Reading data from EEPROM. Start Address: 0x0, Length: 32
Data successfully read from EEPROM.
Data successfully read from EEPROM.

[EEPROM SETTINGS]
System A UID   : 0x1122334455667788
System B UID   : 0x8877665544332211
System C UID   : 0xAABBCCDDEEFF0011
Version          : 0x10001

[Checksum]
 - Stored        : 0x0
 - Computed      : 0x0
Checksum valid.

[EEPROM STRUCTURED HEX DUMP]

[0x00 - 0x07] System A UID
0x0000: 88 77 66 55 44 33 22 11                          |.wfUD3".|

[0x08 - 0x0F] System B UID
0x0008: 11 22 33 44 55 66 77 88                          |."3DUfw.|

[0x10 - 0x17] System C UID
0x0010: 11 00 FF EE DD CC BB AA                          |........|

[0x18 - 0x1B] Version
0x0018: 01 00 01 00                                      |....|

[0x1C - 0x1C] Checksum
0x001C: 00                                               |.|

=====================================
TEST: Sparse Alternating Pattern (0xAA/0x55)
=====================================
Writing sparse alternating pattern...
Writing data to EEPROM. Start Address: 0x0, Length: 64
[WRITE] Chunk 1/3 at 0x000000 - 0x00003F
[#########                     ]  33%  1/3
Writing data to EEPROM. Start Address: 0x10000, Length: 64
[WRITE] Chunk 2/3 at 0x010000 - 0x01003F
[###################           ]  66%  2/3
[... 82 lines cut ...]
    {
      break;
    }

    uint8_t read = _ReadBlock(address, buffer, count);
    if (read == 0)
    {
      break;
    }
    bytes += read;
    address += count;
    buffer += count;
    len -= count;
  }
  return bytes;
}
```

Please excuse me for butchering your code. I hope the code makes some sense to you (and it doesn't give you an aneurysm 😵 )

Thanks again for your time

All the best

dzalf

RobTillaart, 2025-05-28:
Thanks for all this info, need time to digest and process. 
will probably next week due to Ascension day, which is a holiday (long weekend) here.

RobTillaart, 2025-07-05:
Not forgotten, just too busy

RobTillaart, 2025-07-31:
@dzalf 
updated comments and readme.md

RobTillaart, 2025-08-01:
@dzalf 
I merged the develop branch into master and will release it.
If the library does not work, please open an issue under it 

https://github.com/RobTillaart/I2C_CAT24M01

For now I close this issue.

closed 2025-08-01 by RobTillaart

dzalf, 2025-08-01:
Hi @RobTillaart I deeply appreciate your attention to this.

Please give me some time to test it a d I will get back to you.

Best wishes

dzalf

😎

RobTillaart, 2025-08-01:
@dzalf 
Take your time, the library is released (0.1.0) so other people can give their feedback too.