RobTillaart/I2C_EEPROM #83: Single Parameter Constructor Does not Work

TonyRiddiough, 2025-05-29:
Unfortunately the constructor with just the address does not work. After using that constructor, begin and isConnected both return false. Environment: MEGA2560; Arduino IDE 2.3.6; windows10.

RobTillaart, 2025-05-29:
Thanks for reporting here,
Labelled it as bug ad it sounds serious.

RobTillaart, 2025-05-29:
Some questions to get the picture
Which EEPROM do you use?
What address do you use?

I find it strange that isConnected() fails as it is only uses the address.

So my first hypotheses is that the address configured in the code and in the hardware had a mismatch. 
Can you check the hardware connection and run an i2c scanner to verify the address configured?

RobTillaart, 2025-05-30:
Thanks for the additional info.

TonyRiddiough, 2025-06-01:
The current version works fine, sorry for the inconvenience I caused you. 
What I don't understand is that the version I had was the one the Arduino IDE provided only a few days before.

RobTillaart, 2025-06-01:
Good to hear the problem is solved now. 

> What I don't understand is that the version I had was the one the Arduino IDE provided only a few days before.

Sometimes the library manager (?) corrupts the download / fails without an explicit cause. 
Occurred to me maybe a dozen times.

TonyRiddiough, 2025-06-01:
> I2C scanning looks great.
>
> I2C_EEPROM_VERSION: 1.5.2
>
> Version 1.5.2 is from Dec 19, 2021.
> The latest version 1.9.2 is from Nov 25 2024, almost three years newer.
> A lot of fixes and improvements have been added since.
>
> I propose you update the library and redo the tests.

Hi Rob,

   Updated the library and added it to the I2C Scanner. Result now:

I2C Scanner

-----------

0X20    MCP23017 16 Bit Digital Extender(0)

0X21    MCP23017 16 Bit Digital Extender(1)

0X27    PCF8574 8 Bit Digital Extender(7)

0X50    AT24C256 32KB EEPROM(0)

0X57    AT24C32 4KB EEPROM(7)

0X58    Unknown Device

0X68    DS3231 RTC

Thank you again for your help.

Regards,
   Tony Riddiough

TonyRiddiough, 2025-08-26:
Hi Rob,

   I have found that if an instance is created at global scope, the device size is determined incorrectly. The instantiation performs the initialisation, however, at global scope, Wire has not been initialised, hence, the page size defaults to 64. This works fine for the larger eproms, but fails for the smaller ones. I suggest you either insert a warning to this effect or move the code into 'begin'.

Regards,

   Tony Riddiough

RobTillaart, 2025-08-27:
@TonyRiddiough 

Gave it some thoughts, read some code etc, here a summary
- deviceSize is only used to determine the pageSize
- pageSize is used when writing / reading
- so a too large deviceSize, causes an too large pageSize resulting in errors. 
  - **confirms the BUG**
- Adding a warning to the documentation is easy 
  - **ACTION** update constructor text
- alternative might be to remove the "size-less" constructor, so the user has to define it always. 
  - would be a major break so not now, future?
- alternative might be to select the smallest page size possible (I2C_PAGESIZE_24LC01 = 8) as that would allow all devices too work, although the larger ones would slow down.
  - would be a performance break, however it might be accepatble?
- catching an incorrect deviceSize in begin() is only possible if and only if the device is connected and address is right. (see below).
- deviceSize is **NOT** used to prevent access of non existing memory addresses. (now user responsibility, ACTION: make  explicit too in readme.md)
- other observations
  - uint8_t getPageSize(uint32_t deviceSize) should be renamed to **calculatePageSize()** == more correct naming.
- readme.md documentation of begin() is incorrect
  - **ACTION** redo.

----

To catch the size in **begin()** it could be done after the **isConnected()** check 

```cpp
bool I2C_eeprom::begin(int8_t writeProtectPin)
{
  //  if (_wire == 0) SPRNL("zero");  //  test #48
  _lastWrite = 0;
  _writeProtectPin = writeProtectPin;
  if (_writeProtectPin >= 0)
  {
    _autoWriteProtect = EN_AUTO_WRITE_PROTECT;
    pinMode(_writeProtectPin, OUTPUT);
    preventWrite();
  }
  if (isConnected() == false) return false;  //  no device found on given _address
  
  uint32_t deviceSize = determineSize()
  if (deviceSize == 0) return false;  // could not determine...

  _deviceSize = setDeviceSize(deviceSize);
  _pageSize = getPageSize(_deviceSize);
  return true;
}
```

However this construct would always call **determineSize()** which does writes so it will add to number of writes on used locations.

----

Will add this table to the readme.md too, constructor section.

|  Define                  |  bytes  |  Notes  |
|:-------------------------|--------:|:--------|
|  I2C_DEVICESIZE_24LC512  |  65536  |
|  I2C_DEVICESIZE_24LC256  |  32768  |  most used 
|  I2C_DEVICESIZE_24LC128  |  16384  |
|  I2C_DEVICESIZE_24LC64   |   8192  |
|  I2C_DEVICESIZE_24LC32   |   4096  |
|  I2C_DEVICESIZE_24LC16   |   2048  |
|  I2C_DEVICESIZE_24LC08   |   1024  |
|  I2C_DEVICESIZE_24LC04   |    512  |
|  I2C_DEVICESIZE_24LC02   |    256  |
|  I2C_DEVICESIZE_24LC01   |    128  |

----

Conclusion so far => improve the documentation.

RobTillaart, 2025-08-27:
Created a develop branch to update the readme.md

commit a794f6d7 (2025-08-27): fix #83, update readme.md